oliexdev/openScale #908: Mi Scale 2 - Invalid data (10 instead of 13 bytes) is being copied into firstWeight and secondWeight for parsing.

pitus, 2023-01-02:
I was trying to understand the data for **Mi Scale 2** from this code and I noticed that the `firstWeight` and the `secondWeight` copies **use only 10 bytes from the `data[]`** but the `parseBytes()` **expects 13 bytes** and the `data.length` is **2*13=26** which suggests that the `firstWeight` should copy from **0-13** and `secondWeight` from **13-26** - or am I misunderstanding that part of code?

https://github.com/oliexdev/openScale/blob/83c2d941ea8ec349c92fba173398c97a5ee45b70/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothMiScale2.java#L78
https://github.com/oliexdev/openScale/blob/83c2d941ea8ec349c92fba173398c97a5ee45b70/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothMiScale2.java#L79

x11x, 2023-01-16:
I noticed this too. It looks like the code was copied from V1 and not updated (V1 uses 10 byte messages for weight measurements). This suggests V2 does not send multiple measurements in one message, otherwise someone would have reported incorrect readings. So I guess this is dead code? (just a guess, hard to be sure unless someone finds a situation when the scale sends 26 bytes here).

x11x, 2023-01-16:
Experimenting on my scale "Body Composition Scale 2", it seems to send single 13 byte measurements at a time, I have not seen it send 26 byte messages

oliexdev, 2023-01-20:
@x11x is right the initial v2 code was copied from v1. So it could be dead code. Unfortunately, I don't own any mi scale anymore. At least in the v1 of the scale sometimes a 20 bytes with two combined measurement was send from the scale (If I remember right, it was while in the advertisement data not in the history data). For the v2 I am not sure, if I ever saw it.

I have removed this check in the source code for now.

closed 2023-01-20 by oliexdev