oliexdev/openScale #910: Crash when accessing bluetooth

d33psky, 2023-01-10:
**Describe the bug**
An unexpected error handler with below stack trace shows during the initial setup when adding a bluetooth scale, as well as after a 'restart app' in settings->bluetooth and click - to add 'your bluetooth scale'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings->bluetooth
2. Click on '-' to add a scale
3. See error

Reproduced with v2.4.6 (61) from F-Droid

**Expected behavior**
Expect to see a list of bluetooth devices

**Additional context**
Bluetooth access is set to 'always ask' in settings->apps->manage apps->permissions->other permissions->permissions->'change bluetooth connectivity'.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
```
2023-01-10 15:04:35.554 Debug [2] AboutPreferences: Debug log enabled, openScale v2.4.6 (61), SDK 31, Xiaomi M2101K6G
2023-01-10 15:04:35.560 Debug [2] AboutPreferences: Selected user id(1) name(Hans) birthday(Sun Jul 30 00:00:00 GMT+01:00 1967) age(55) body height(180.00) scale unit(kg) gender(male) initial weight(68.00) goal weight(68.00) goal date(Mon Jul 10 14:19:23 GMT+02:00 2023) measure unt(cm) activity level(1) assisted weighing(false)
```
And the backtrace : 
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-10 14:23:56 
Device: Xiaomi M2101K6G 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{5e9c42f 2393:com.health.openscale/u0a398} (pid=2393, uid=10398) requires android.permission.BLUETOOTH_CONNECT
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2426)
	at android.os.Parcel.createException(Parcel.java:2410)
	at android.os.Parcel.readException(Parcel.java:2393)
	at android.os.Parcel.readException(Parcel.java:2335)
	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2626)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1745)
	at android.app.Activity.startActivityForResult(Activity.java:5507)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:597)
	at android.app.Activity.startActivityForResult(Activity.java:5465)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:583)
	at com.health.openscale.gui.utils.PermissionHelper.requestBluetoothPermission(PermissionHelper.java:56)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:102)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:771)
	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:690)
	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:855)
	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:154)
	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:123)
	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:306)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1853)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1764)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1701)
	at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:488)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:210)
	at android.os.Looper.loop(Looper.java:299)
	at android.app.ActivityThread.main(ActivityThread.java:8168)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1037)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1139)
	at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:1058)
	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:724)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1301)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1273)
```

d33psky, 2023-01-10:
Good news: this is just a disabled bluetooth in phone->settings->bluetooth .
Might want to add this case with bluetooth permission is denied with a 'please enable bluetooth' message.

oliexdev, 2023-01-20:
I rewrote the permission handling to the latest Android methods, should be fixed in the latest dev version

closed 2023-01-20 by oliexdev