oliexdev/openScale #913: Crash when modify user

Mikolaakos, 2023-01-19:
In list view I edit a measurement, change user. When tap save icon in upper right corner, the app crashes. 

Build version: 2.4.6-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-19 22:41:12 
Device: Google Pixel 6 Pro 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime (code 2067 SQLITE_CONSTRAINT_UNIQUE)
	at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
	at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(SQLiteConnection.java:913)
	at android.database.sqlite.SQLiteSession.executeForChangedRowCount(SQLiteSession.java:756)
	at android.database.sqlite.SQLiteStatement.executeUpdateDelete(SQLiteStatement.java:67)
	at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeUpdateDelete(FrameworkSQLiteStatement.java:46)
	at androidx.room.EntityDeletionOrUpdateAdapter.handle(EntityDeletionOrUpdateAdapter.java:70)
	at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.update(ScaleMeasurementDAO_Impl.java:176)
	at com.health.openscale.core.OpenScale.updateScaleMeasurement(OpenScale.java:416)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.saveScaleData(MeasurementEntryFragment.java:346)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.onOptionsItemSelected(MeasurementEntryFragment.java:200)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3154)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:2937)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3158)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:2937)
	at androidx.fragment.app.FragmentController.dispatchOptionsItemSelected(FragmentController.java:427)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:334)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:264)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:221)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:834)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:985)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:975)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
	at android.view.View.performClick(View.java:7506)
	at android.view.View.performClickInternal(View.java:7483)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:29341)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7872)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)

oliexdev, 2023-01-20:
It seems you have two users with the same user id which shouldn't be possible. Did you import something? Could you do a step by step approach from a clean openScale installation so I can reproduce it?

Mikolaakos, 2023-01-22:
Hi OliE,

This is my second or third scale, all different brands, actually I don't
remember if I imported my data - probably I did.

There are three users in the app, it seems that the measures are mixed up
among them. I attach the exported data.

Shall I delete all data of the app, create users, import my data, and move
the specific measures between them?

Üdv,
Ákos

OliE ***@***.***> ezt írta (időpont: 2023. jan. 20., P,
17:04):

oliexdev, 2023-01-22:
I can't see your exported data. Did you export your data as a csv file or db?

Mikolaakos, 2023-01-22:
Three csv were attached to my email, maybe the github server deleted them.

oliexdev, 2023-01-28:
please try to attach it via the web interface of github

Mikolaakos, 2023-01-30:
[openScale Hanna.csv](https://github.com/oliexdev/openScale/files/10540914/openScale.Hanna.csv)
[openScale Brigitta .csv](https://github.com/oliexdev/openScale/files/10540915/openScale.Brigitta.csv)
[openScale Mikola.akos.csv](https://github.com/oliexdev/openScale/files/10540916/openScale.Mikola.akos.csv)
here they are

oliexdev, 2023-01-31:
I don't have an issue here. I could import all three .csv files into openScale without an error.

1.First you need to create three users (Hanna, Brigitta and Mikola) 
2. Select Hanna then go to upper right the three dots -> click on import -> then select openScale Hanna.csv
3. Select Brigitta then go to upper right the three dots -> click on import -> then select openScale Brigitta.csv
4. Select Mikola then go to upper right the three dots -> click on import -> then select openScale Mikola.csv

On which step do you get an error?

closed 2023-02-12 by oliexdev

Strubbl, 2024-12-20:
> In list view I edit a measurement, change user. When tap save icon in upper right corner, the app crashes.

I can reproduce this crash with these steps.

Strubbl, 2024-12-20:
The crash is reproducible. I get almost the same exception:

```
android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime (code 2067 SQLITE_CONSTRAINT_UNIQUE[2067])
	at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
	at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(SQLiteConnection.java:1241)
	at android.database.sqlite.SQLiteSession.executeForChangedRowCount(SQLiteSession.java:756)
	at android.database.sqlite.SQLiteStatement.executeUpdateDelete(SQLiteStatement.java:67)
	at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeUpdateDelete(FrameworkSQLiteStatement.kt:38)
	at androidx.room.EntityDeletionOrUpdateAdapter.handle(EntityDeletionOrUpdateAdapter.kt:61)
	at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.update(ScaleMeasurementDAO_Impl.java:180)
	at com.health.openscale.core.OpenScale.updateScaleMeasurement(OpenScale.java:416)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.saveScaleData(MeasurementEntryFragment.java:346)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.onOptionsItemSelected(MeasurementEntryFragment.java:200)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3284)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:3168)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3288)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:3168)
	at androidx.fragment.app.FragmentManager$2.onMenuItemSelected(FragmentManager.java:503)
	at androidx.core.view.MenuHostHelper.onMenuItemSelected(MenuHostHelper.java:107)
	at androidx.activity.ComponentActivity.onMenuItemSelected(ComponentActivity.java:532)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:264)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:269)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:224)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:769)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:833)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:157)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:984)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:974)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:620)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:155)
	at android.view.View.performClick(View.java:8055)
	at android.widget.TextView.performClick(TextView.java:17819)
	at android.view.View.performClickInternal(View.java:8032)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:31923)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:230)
	at android.os.Looper.loop(Looper.java:319)
	at android.app.ActivityThread.main(ActivityThread.java:9063)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:588)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
```

Shall i open a new issue or can we reopen this one?

Strubbl, 2024-12-20:
This issue is also reproducible in the sqlite itself. If i execute `update "main"."scaleMeasurements" SET userId='3' where "id"='566';`, i also get the same SQL error `UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime`.

I do not understand this, because a measurement with the exactly same time does not exist for any user.