oliexdev/openScale #918: openscale sync feature request

ryan77627, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
Hello, hopefully this is an acceptable place to file a feature request for the wger sync add-on in Google play. I'd like to sync my weight to a self hosted wger instance, however I cannot since the app disallows HTTP connections and ignores my custom root CA in the user cert store

**Describe the solution you'd like**
Preferably allow custom certs to be accepted by changing the app's manifest (I believe that's all that needs to be done, but I'm not an android app dev so I could be wrong)

**Describe alternatives you've considered**
None really, I don't think there is an alternative short of hand entering my weight

**Additional context**
Error on HTTP connection:
```
2023-02-02 01:35:09.146 Error [2] WgerSync$g: get connection failure CLEARTEXT communication to wger.myserver.home.arpa not permitted by network security policy
```

Error on HTTPS connection:
```
2023-02-02 01:36:14.303 Error [2] WgerSync$g: get connection failure java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.
```

Focshole, 2023-02-03:
> 2023-02-02 01:36:14.303 Error [2] WgerSync$g: get connection failure java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.

This one is due to a self signed certificate I'd guess. I'd guess you'd have to add it as a reliable certificate manually in android. Maybe[ something like this](https://medium.com/hackers-secrets/adding-a-certificate-to-android-system-trust-store-ae8ca3519a85).

Also, I cannot find any reference about this wger app. Could you please provide a link?

ryan77627, 2023-02-03:
Hello, yes it is self signed, but the CA has been added to the user store (firefox for example sees it as trusted if I turn on the setting to allow the user cert store). However, as of Android 11, separation between the two stores is enforced and an app needs to specifically opt in to trusting these certificates (or allowing plaintext communication apparently). Also, I'm referring to the app [found here](https://github.com/oliexdev/openScale/wiki/openScale-sync)

oliexdev, 2023-02-04:
it seems the ERROR on HTTP connection, could be resolved with adding `android:usesCleartextTraffic="true"` into the <application> manifast, see https://stackoverflow.com/questions/60175852/cleartext-communication-not-permitted-by-network-security-policy-working-on-my-m The easiest way.

and the second one, could be easily resolved to ignore any ca certificates which is of course unsecure, https://futurestud.io/tutorials/retrofit-2-how-to-trust-unsafe-ssl-certificates-self-signed-expired
It's more complex and insecure.

The best way is of course to add those self-signed ca certifcates into the app itself but that its not possible because I need to know them before I release a new version, see https://developer.android.com/training/articles/security-config

So, if it's ok for your to have not a secure connection I could add the usesCleartextTraffic into the manifast.

closed 2025-06-25 by oliexdev