oliexdev/openScale #919: Support for Beurer BY90

joergi, 2023-02-03:
**Scale name**
Beurer BY90
https://www.beurer.com/web/gb/products/babycare/baby-scale/by-90.php

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

I used the B[eurer BabyCare app](https://play.google.com/store/apps/details?id=com.beurer.connect.babycare) for getting the data

1. Bluetooth HCI Snoop log file

user settings in the vendors app:
```
sex: female
body height: 50cm, 
age: 01.01.2022, 
activity level ? (nothing)
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
weight: 6.120kg
date: 16:21 
water, muscle , fat etc. (nothing like this is in the app)
```
[btsnoop_hci_1-6dot120_1621.log](https://github.com/oliexdev/openScale/files/10580659/btsnoop_hci_1-6dot120_1621.log)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex: female
body height: 50cm, 
age: 01.01.2022, 
activity level ? (nothing)

```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
weight: 6.875kg
date: 16:39 
water, muscle , fat etc. (nothing like this is in the app)

```
[btsnoop_hci_2_6dot875_1639.log](https://github.com/oliexdev/openScale/files/10580708/btsnoop_hci_2_6dot875_1639.log)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex: female
body height: 50cm, 
age: 01.01.2022, 
activity level ? (nothing)

```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
weight: 8.230kg
date: 16:50 
water, muscle , fat etc. (nothing like this is in the app)

```
[btsnoop_hci_3_8dot230_1650.log](https://github.com/oliexdev/openScale/files/10580719/btsnoop_hci_3_8dot230_1650.log)

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

I have done it, but where can I find it? it's not in the description. will add it if you can tell me, where it is. thanks

_--> Attach the openScale debug log file here <--_

joergi, 2023-02-03:
if you need any other information, please let me know.

oliexdev, 2023-02-04:
@joergi you could use the app https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner to discover the Bluetooth services and characteristic make some screenshots

joergi, 2023-02-07:
Please let me know, if you need more information... 

[image]

[image]

[image]

![Screenshot_20230208-002815.png](https://user-im[image]

24e2-5e28-4f7e-9a05-f62f5961a11d.png)

[image]

joergi, 2023-02-17:
@oliexdev have you been able to have a look if the data I posted are good?
Can I provide any other information?
(I was already looking into the logs but can't really help here...)

oliexdev, 2023-02-18:
@joergi your HCI log files doesn't cover the Bluetooth weight process. It has unfortunately many other Bluetooth packets like Sonos or Apple. So you have some other Bluetooth communications.
But from your screenshots your scale should use the standard weight bluetooth profile. Please try out the latest dev-build --> https://github.com/oliexdev/openScale/releases/tag/dev-build 
Please also enable the openScale log file while your using the openScale app and post here this log file.