oliexdev/openScale #923: Incorrect time of the measurement

woj-tek, 2023-02-10:
**Describe the bug**
For some reason, starting yesterday, OpenScale displays incorrect timestamp of the measurement. For example measurement took at 7:36 would be displayed with timestamp 10:36. 
I'm currently in CLST timezone, which is `UTC -3` but I've been here for a couple of months now. There haven't been any update made (that I recall of). Zepp life was update on 26th of Jan so long before the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable bluetooth in the OS
2. step on the scale
3. get synchronised measurement with incorrect timestamp

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): not tried.

**Debug log**

(Measurement took at 8:17, displays 11:17)
[openScale_2023-02-10_08-16.txt](https://github.com/oliexdev/openScale/files/10706815/openScale_2023-02-10_08-16.txt)

Relevant line:
```
2023-02-10 08:17:20.714 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Fri Feb 10 11:17:00 GMT-03:00 2023, WEIGHT: 75,40, FAT: 22,64, WATER: 53,07, MUSCLE: 73,42, LBM: 0,00, WAIST: 0,00, HIP: 0,00, BONE: 2,80, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
```

oliexdev, 2023-02-11:
What do you mean by you are currently in CLST timezone? Did it happen after you moved in this timezone? The current time/date is set on your scale with your date/time set from your smartphone.

woj-tek, 2023-02-11:
I moved to CLST timezone 2 months ago, I got new mi scale, added it to openScale and everything was working just fine this past two months. 2 days ago all of the sudden the timestamp started to not match actual time of the measurement (measurement at 7:30 woild be logged with timestamp 10:30). To the best of my recollection nothing has changed.

I checked zepp application but I don't see any setting related to timezone of the scale. Nor did I saw it in the openScale settings.

oliexdev, 2023-02-27:
The time/date is set every time your connect to the scale in the background. The reference of the date/time is your setting in your smartphone (main date/time)! Please check if your smartphone has the correct time/date. You could also try to reset your scale, please look into the scale manual how you do that.

woj-tek, 2023-02-27:
Weird. I check Zapp app and the timestamp in it was correct. But I went ahead with factory reset and... it seems to have worked :-)

oliexdev, 2023-02-27:
ok great, I won't recommend to use the vendor app and openScale in parallel because some scales needs to be set an internal key to work. If you switch between two apps then the scale "get confused".

closed 2023-02-27 by oliexdev

woj-tek, 2023-02-27:
I would just use openScale, but... I need official app for Mi Band to work (Notify for Mi Band, even though it's 3rd party still requires it to work for some reason...)

oliexdev, 2023-02-28:
@woj-tek you could try https://github.com/Freeyourgadget/Gadgetbridge for your Mi Band.

woj-tek, 2023-03-01:
@oliexdev oh, thanks! Looks promising. I wonder if it would work with SleepAsAndroid though, will test. Thanks a million! :-)