oliexdev/openScale #925: Support Lepulse Lescale P1

H89P, 2023-02-12:
Lepulse Lescale P1
https://lepulsefit.com/products/lescale-p1-professional-8-electrodes-body-fat-scale

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:

[image]

```
measured true values in the vendors app for the 1. HCI Snoop log file:
This is the complete report from measurement 1
```
[Lescale_P1_20230212173219_0.pdf](https://github.com/oliexdev/openScale/files/10716594/Lescale_P1_20230212173219_0.pdf)

[btsnooz_hci.log](https://github.com/oliexdev/openScale/files/10716694/btsnooz_hci.log)

2. Bluetooth HCI Snoop log file
[image]

```
measured true values in the vendors app for the 2. HCI Snoop log file:
This is the complete report from measurement 
```
[Lescale_P1_20230212181653_0.pdf](https://github.com/oliexdev/openScale/files/10716720/Lescale_P1_20230212181653_0.pdf)

[btsnooz_hci.log](https://github.com/oliexdev/openScale/files/10716730/btsnooz_hci.log)

_--> Attach the 2. HCI Snoop log file here <--_

3. Bluetooth HCI Snoop log file
user settings in the vendors app:

[image]

```
measured true values in the vendors app for the 3. HCI Snoop log file:
This is the complete report from measurement 3
```
[Lescale_P1_20230212182229_0.pdf](https://github.com/oliexdev/openScale/files/10716733/Lescale_P1_20230212182229_0.pdf)

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/10716740/btsnoop_hci.log)

[openScale_2023-02-12_17-01.txt](https://github.com/oliexdev/openScale/files/10716734/openScale_2023-02-12_17-01.txt)

[image]

H89P, 2023-02-12:
If you need anything else let me know

mauricecyril, 2023-03-22:
Also noticed that there are few other scales with similar 8 electrodes. 

Posture FG-2001-1 
FIAR Smart Body Fat Scale

They seem to use an app called Fitdays or Starfit developed by icomon

https://play.google.com/store/apps/details?id=com.icomon.starfit.starfit

mauricecyril, 2023-03-22:
Hi @oliexdev was wondering if there are any other items you need from @H89P to see if the Lepulse Lescale P1 can be integrated?

ergunucman, 2023-08-22:
Hi, I also have the Lepulse lescale P1 device. through reverse engineering i am able to listen in bluetooth packets between mobile app and scale. but I don't understand data traffic as I don't have bluetooth protocols. Please help me.
Let me give an example of packets between two devices:
weight measurement only. kilo.73.2
85 07 00 a2 03 19 01 1d b4 0000000000000000000010
During the weight measurement, the package number starting with 85 is incremented each time. But the command 07 00 a2 03 19 01 has not changed at all. In this measurement, 1d b4 defines the weight value, but I cannot match the 1d b4 (hex) to 73.2 (dec) weight value. Where do you think I am going wrong?