oliexdev/openScale #926: Connection issue on new Android 13 phone

Tea-Ice, 2023-02-19:
**Describe the bug**
Whenever I try to connect to the BF710, the app says connected, but the scale no longer shows the Bluetooth icon.
It used to work fine on my older phone, and if I use HealthManager Pro it works, but normal HealthManager also fails to connect.

**To Reproduce**
Try to connect the Samsung S23U to the 

Beurer BF710

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Yes

**Expected behavior**
The connection to form properly

**Additional context**
I used to have a S21U, and it still worked perfectly even while on Android 13. Really confused about this. I have also tried resetting the mobile app, the scale, and restarting my phone, HealthManager Pro works immediately and without any problems, except that my data is being sent to who knows how many servers.

**Debug log**
The debug log was made after uninstalling the 3rd party apps, and resetting my network settings(Bluetooth, Wi-Fi, Mobile) in the system menu.

[openScale_2023-02-19_10-19.txt](https://github.com/oliexdev/openScale/files/10776439/openScale_2023-02-19_10-19.txt)

oliexdev, 2023-02-26:
maybe your scale batteries are low?

gostomek100, 2023-02-26:
Hello,
I have similar problem.
I got Samsung S23+ with new Android 13 and Sanitas SBF70 and also message is that connection is correct and  data is not synchronized.

fischy667, 2023-05-02:
Same here with a Samsung Galaxy S23.

palmtop, 2023-07-12:
I have a very similar problem with my Yunmai Mini scale and Samsung Galaxy S10 and Galaxy S22. Earlier everything was working almost without problems, now most of the time the connection to the scale succeds, but the results are not transferred back. If I remove the batteries from the scale, I can make a sucessfull measurement, but the next is again not sucessfull.

It the failing case from the log it can be seen, that sending the user data is sucessfull, but sending the date is failing, and when I stand on the sacle, then it is already disconnected.

I wrote a python script to test, but it is always sucessfull, even, when sending the same data as openScale is sending.

I think that at least in the UI a warning should be displayed, when writing to the scale is not susessfull.

I attach the logs for the two cases, and the output from the python program, where most packets are dumped.

[openScale_2023-07-11_19-11_sucessfull.txt](https://github.com/oliexdev/openScale/files/12031275/openScale_2023-07-11_19-11_sucessfull.txt)
[openScale_2023-07-11_06-14_failing.txt](https://github.com/oliexdev/openScale/files/12031276/openScale_2023-07-11_06-14_failing.txt)
[python output.txt](https://github.com/oliexdev/openScale/files/12031289/python.output.txt)

palmtop, 2023-07-23:
This seems to be a timing related bug (at least for Yunmai Mini). 

I have added 400 ms sleep to the step 0 in BluetoothYunmaiSE_Mini.java/onNextStep, and since it always reads the data.

Then I compiled the app and sterted to use. There is one strange thingm that is is recording weigth ~0,6 kg more than the measurement. From the logs I see that the measurement is correct, but after the calculation it gets modified.

```
2023-07-23 08:49:51.044 Debug [2] BluetoothYunmaiSE_Mini: Extract the fat value from received bytes
2023-07-23 08:49:51.052 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 64 BC CD 8E 00 00 58 57 20 A4 00 96 0A 3C B9]
2023-07-23 08:49:51.055 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 83,56, fat: 26,20, resistance: 150]
2023-07-23 08:49:51.058 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(Lacó) birthday(Sun Oct 01 00:00:00 GMT+01:00 1967) age(55) body height(178,00) scale unit(kg) gender(male) initial weight(84,00) goal enabled(false) goal weight(-1,00) goal date(Tue Jul 18 21:28:33 GMT+02:00 2023) measure unt(cm) activity level(1) assisted weighing(false)]
2023-07-23 08:49:51.062 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Sun Jul 23 08:49:50 GMT+02:00 2023, WEIGHT: 83,56, FAT: 26,20, WATER: 53,58, MUSCLE: 49,45, LBM: 61,67, WAIST: 0,00, HIP: 0,00, BONE: 3,25, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: ]
2023-07-23 08:49:51.086 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Sun Jul 23 08:49:50 GMT+02:00 2023, WEIGHT: 84,23, FAT: 26,20, WATER: 53,58, MUSCLE: 49,45, LBM: 61,67, WAIST: 0,00, HIP: 0,00, BONE: 3,25, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
```
You can see that the measured weight is 83,56 Kg and the added measurement is 84,23

@oliexdev 

I would appresiate if the delay would be added to the official sources, this weight mismatch figured out (may be I have some wrong settings?), and a new pro version would be published.

fischy667, 2023-08-19:
I'm now on Android 14 Beta on my Galaxy S23. Nothing has changed regarding the Bluetooth connection problem.

palmtop, 2023-08-19:
@fischy667: What scale are you using?
I did the correction only for the Yunmai, there the problem was, that the new phones are so fast, that they send the second bluetooth command to the scale, before thay are able to process the first one.

fischy667, 2023-08-19:
Sanitas SBF70

palmtop, 2023-08-19:
Oh, I thought, that you have compiled the fix I uploaded, and that was not working for you.
I have looked a bit into the code for the SBF70 but is is very different from the Yunmai code. My change will not help you, and I could not find a simmilar fix for SBF70. 
I am sorry for it. Based on your description I thought that it will be a very simmilar issue.

thegreekgeek, 2024-05-12:
Yeah I'm pretty sure this is the issue I'm having too.

oliexdev, 2024-05-20:
seems fixed with https://github.com/oliexdev/openScale/pull/983

closed 2024-05-20 by oliexdev