oliexdev/openScale #929: Color meanings in the app

vladapalijan, 2023-02-21:
Hello i hope this makes sense, English is not my main language.

 I am interested in what each of the colors mean in the app. 
For instance for the weight i know red is bad but when i gain muscle mass it still shows red arrow like its a bad thing and overall muscle mass is in color red. 

Hope someone who knows can answer me.
Thanks in advance!

oliexdev, 2023-02-27:
could you add screenshots to understand it better what you mean?

vladapalijan, 2023-02-27:
> could you add screenshots to understand it better what you mean?

Ye sure. On the app the colors go blue -> green -> red for everything.

But i think for muscle the colors shouldnt be the same as for weight. Because if you have a lot of muscle mass thats good and not bad as for weight. (color: red).

I think the colors for muscle mass should end with the last one on the right being green instead of the red that is now.

I hope you understood me.

[image]

oliexdev, 2023-02-27:
these are evaluation indicators to see the average value in your specific group (sex, age).
For muscle it is based on the scientific paper  "Skeletal muscle mass and distribution in 468 men and women aged 18–88 yr" by IAN JANSSEN, STEVEN B. HEYMSFIELD, ZIMIAN WANG, and ROBERT ROS in J Appl Physiol89: 81–88, 2000

or read for example https://www.propellolife.com/blogs/news/what-is-a-good-lean-muscle-percentage

Sure if you are body builder you want a lot of muscle mass but that's not the "average healthy" body goal.