oliexdev/openScale #932: Crash since kast Update

amarradi, 2023-02-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Build version: 2.5.0 
Build date: 1981-01-01 01:01:02 
Current date: 2023-02-21 15:58:12 
Device: Motorola moto g(6) plus 
OS version: Android 9 (SDK 28) 
 
Stack trace:  
java.lang.RuntimeException: Unable to start receiver com.health.openscale.core.alarm.ReminderBootReceiver: java.lang.IllegalArgumentException: Invalid URI: openScale backup
	at android.app.ActivityThread.handleReceiver(ActivityThread.java:3605)
	at android.app.ActivityThread.access$1300(ActivityThread.java:213)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1818)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6923)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:870)
Caused by: java.lang.IllegalArgumentException: Invalid URI: openScale backup
	at android.provider.DocumentsContract.getTreeDocumentId(DocumentsContract.java:1014)
	at androidx.documentfile.provider.DocumentFile.fromTreeUri(DocumentFile.java:138)
	at com.health.openscale.core.alarm.AlarmBackupHandler.executeBackup(AlarmBackupHandler.java:100)
	at com.health.openscale.core.alarm.ReminderBootReceiver.handleBackupAlarm(ReminderBootReceiver.java:44)
	at com.health.openscale.core.alarm.ReminderBootReceiver.onReceive(ReminderBootReceiver.java:30)
	at android.app.ActivityThread.handleReceiver(ActivityThread.java:3596)
	... 8 more

amarradi, 2023-02-23:
Here is the Crash dump from the last crash

2023-02-21 16:08:23.140 Debug [2] AboutPreferences: Debug log enabled, openScale v2.5.0 (62), SDK 28, motorola moto g(6) plus
2023-02-21 16:08:23.154 Debug [2] AboutPreferences: Selected user id(1) name(Marcus) birthday(Tue May 25 00:00:00 GMT+02:00 1982) age(40) body height(188,00) scale unit(kg) gender(male) initial weight(94,30) goal enabled(false) goal weight(94,00) goal date(Thu Dec 23 00:00:00 GMT+01:00 2021) measure unt(cm) activity level(1) assisted weighing(false)

oliexdev, 2023-02-26:
Did you update openScale from a previous version or did it happen after a fresh installation? Could you set again the backup folder in the settings and see if this crash happens again?

amarradi, 2023-02-26:
Hi,

I installed it from f-droid and also updated it via f-droid. I just made a
backup of the settings and my data and before your message already removed
the app, also deleted all data and reinstalled to test it.

Mein Krebstagebuch  https://www.marcusradisch.de

oliexdev, 2023-02-27:
and what was the result of your testing? Do you still get the error message after a fresh installation?

amarradi, 2023-02-27:
After uninstalling and installing, I would still have to manually delete my
data again, since nothing seems to have changed. Now I have imported my
data and settings and have not experienced a crash.

Mein Krebstagebuch  https://www.marcusradisch.de

oliexdev, 2023-02-27:
thanks your error should be fixed with the latest dev build. The problem was the old stored internal backup directory.

closed 2023-02-27 by oliexdev