oliexdev/openScale #935: An unexpected error occurred

ToriSong, 2023-02-28:
Build version: 2.5.1-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2023-02-28 23:40:28 
Device: Samsung SM-S901B 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
	at com.health.openscale.gui.table.StickyHeaderTableView.updateMaxWidthHeightOfCell(StickyHeaderTableView.java:611)
	at com.health.openscale.gui.table.StickyHeaderTableView.onMeasure(StickyHeaderTableView.java:421)
	at android.view.View.measure(View.java:27720)
	at androidx.constraintlayout.widget.ConstraintLayout$Measurer.measure(ConstraintLayout.java:811)
	at androidx.constraintlayout.core.widgets.ConstraintWidgetContainer.measure(ConstraintWidgetContainer.java:632)
	at androidx.constraintlayout.core.widgets.analyzer.Direct.verticalSolvingPass(Direct.java:452)
	at androidx.constraintlayout.core.widgets.analyzer.Direct.solveVerticalMatchConstraint(Direct.java:750)
	at androidx.constraintlayout.core.widgets.analyzer.Direct.verticalSolvingPass(Direct.java:503)
	at androidx.constraintlayout.core.widgets.analyzer.Direct.solvingPass(Direct.java:224)
	at androidx.constraintlayout.core.widgets.ConstraintWidgetContainer.layout(ConstraintWidgetContainer.java:693)
	at androidx.constraintlayout.core.widgets.analyzer.BasicMeasure.solveLinearSystem(BasicMeasure.java:160)
	at androidx.constraintlayout.core.widgets.analyzer.BasicMeasure.solverMeasure(BasicMeasure.java:291)
	at androidx.constraintlayout.core.widgets.ConstraintWidgetContainer.measure(ConstraintWidgetContainer.java:120)
	at androidx.constraintlayout.widget.ConstraintLayout.resolveSystem(ConstraintLayout.java:1594)
	at androidx.constraintlayout.widget.ConstraintLayout.onMeasure(ConstraintLayout.java:1708)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1552)
	at android.widget.LinearLayout.measureVertical(LinearLayout.java:842)
	at android.widget.LinearLayout.onMeasure(LinearLayout.java:721)
	at android.view.View.measure(View.java:27720)
	at androidx.drawerlayout.widget.DrawerLayout.onMeasure(DrawerLayout.java:1156)
	at android.view.View.measure(View.java:27720)
	at android.widget.RelativeLayout.measureChildHorizontal(RelativeLayout.java:735)
	at android.widget.RelativeLayout.onMeasure(RelativeLayout.java:481)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
	at androidx.appcompat.widget.ContentFrameLayout.onMeasure(ContentFrameLayout.java:145)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1552)
	at android.widget.LinearLayout.measureVertical(LinearLayout.java:842)
	at android.widget.LinearLayout.onMeasure(LinearLayout.java:721)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1552)
	at android.widget.LinearLayout.measureVertical(LinearLayout.java:842)
	at android.widget.LinearLayout.onMeasure(LinearLayout.java:721)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:7352)
	at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
	at com.android.internal.policy.DecorView.onMeasure(DecorView.java:1375)
	at android.view.View.measure(View.java:27720)
	at android.view.ViewRootImpl.performMeasure(ViewRootImpl.java:4841)
	at android.view.ViewRootImpl.measureHierarchy(ViewRootImpl.java:3426)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:3732)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:3116)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:10885)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1301)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1309)
	at android.view.Choreographer.doCallbacks(Choreographer.java:923)
	at android.view.Choreographer.doFrame(Choreographer.java:852)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1283)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8757)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)

oliexdev, 2023-03-02:
I can't reproduce this error. Which steps are necessary to reproduce this error? Could you send me your database (Settings->Backup->Export Backup)?

vlk12345, 2023-03-02:
It is necessary only start application. I cannot export backup because app immediately crash.

vlk12345, 2023-03-28:
Even reinstall did not help.

PR #1047 (2024-05-20): (fix) StickyHeaderTableView: crash on certain strings
This PR handle the case where some text ends by an `\n` (ex: 'something\n'). From `String.split` docs, ` trailing empty strings will be discarded.`. That cause the exception since there are less element than expected in the resulting array. 

From this screenshot, you can see the bug in action.
[image]

Thanks for the app oliexdev :)

Fix #935
closed 2024-05-20 by oliexdev