oliexdev/openScale #936: Widget for date entry has missing/invisible OK Button

DamageIncorporated, 2023-03-04:
Whe. I set the target date for my future weight today, I was presented a date entry widget, but couldn't see the OK Button to actually set the date.

Whet I tapped the lower right corner, the date was saved, so I assume the button is present, but invisible.

This happened to me on a de-googled device (e-os ) and in dark mode.

Maybe this is enough Info to reproduce this issue?

I have to go now, but will provide further details later when needed...

Tzatzikiiii, 2023-03-09:
Can confirm. Also on dark mode. However the button is not invisible, the font is just black so it's really hard to read. Same thing with the "Cancel" button to the left.

oliexdev, 2023-07-22:
Sorry yeah that's true, I will close this one please see #944

closed 2023-07-22 by oliexdev