oliexdev/openScale #937: Bluetooth requires location to be turned on (Mi Smart Scale 2 / OnePlus NordCE 5G)

ohsotech, 2023-03-06:
Mi Smart Scale 2 requires location to be enabled on OnePlus NordCE 5G. This was not the case a few weeks ago.

If I press the bluetooth icon (or have it enabled to auto turn on) it shows this screen, opens the 'enable location' screen and doesn't turn on bluetooth unless location is already on.

[image]

This should only be required for *finding* a bluetooth scale for the first time, no? If I click on my already-paired scale in the settings, it opens the same window, with the small difference that it has a 'no' button too.

So seems like a bug?

FAQ info which I believe was mentioned in https://github.com/oliexdev/openScale/issues/887 :

> **Why do openScale needs permission to the coarse location?**
> The answer is simple because the Android API >= 6.0 needs access to the coarse location to search for Bluetooth devices. The App openScale doesn't read your location at any time. Read the official [Android Bluetooth Developer Guide](https://developer.android.com/guide/topics/connectivity/bluetooth-le.html#permissions) or read the [Android 6.0 Changelog](https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html#behavior-hardware-id) for a more technical description.

Doesn't really make sense to me. Especially because in android app settings, location is not allowed for openscale. And if I enable location permission, the exact same thing happens - unless it's actually turned on.

suncoastkid, 2023-03-08:
Same issue on LineageOS Redfin device. Very disappointing.

0xDB6, 2023-09-10:
Bump. Same issue here on a Samsung device with Android 13. I vaguely remember this having been working before.