oliexdev/openScale #940: Add SBF73 its same SBF72

drsgdfnhtyjktyjt, 2023-03-12:
Hello, I just bought a scale SBF73 it and not support, but seen that the SBF72 is well recognized I tested by changing just the SBF72 in SBF73 and it works very well, the user already created on the scale is well recognized and all the information on the body are also.

if you need pictures of the app said the month, for the moment I will keep the one I created to support the SBF73

I don't know how to extract the log ble and sorry for my english I translate them by deepl

2spirit, 2023-06-02:
Hello, where do you change the name SBF72 -> SBF73 ?

2spirit, 2023-06-02:
OK, I found it , @oliexdev , in ./android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothFactory.java
line 130 , you can add "deviceName.equals("SBF73")"

Thanks !

closed 2023-07-22 by oliexdev