oliexdev/openScale #941: Only connecting to Bluetooth for a few seconds

TheHornedJester, 2023-03-14:
**Describe the bug**
I have been using openScale for over 5 years. The problem is only with my new Galaxy S23 Ultra, my previous phone was a Note 9, and is still working fine.
Bluetooth connects with my YUNMAI-SIGNAL-MW for about 4 seconds, then the bluetooth icon disappears from my scales and no measurements are recorded. Except for 2 occasions. I happened to capture one of those occasions in the log file.

**To Reproduce**
1. Open app.
2. Step on scales.
3. Wait for nothing to happen.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 
v2.5.1-pro (63)

**Expected behavior**
Record a measurement.

**Additional context**
You will find a failure and a success in the log file. Line 39 maybe of interest.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
[openScale_2023-03-10_13-55 KEEP Has the data to sort out Scale problem.txt](https://github.com/oliexdev/openScale/files/10964800/openScale_2023-03-10_13-55.KEEP.Has.the.data.to.sort.out.Scale.problem.txt)

oliexdev, 2023-07-22:
It seems some commands couldn't be send to the scale. Could you check if the batteries are low? Could be also a problem with the BLE implementation from your smartphone.

closed 2023-07-22 by oliexdev

TheHornedJester, 2023-07-29:
Hi Olie,

Thanks for getting back to me.

It appears the problem was Samsungs. I have tested it after receiving an
operating system update and it is now working fine.

Thanks for all your effort in creating a good app that stands the test of
time.
Regards,

John