oliexdev/openScale #943: openScale no longer recording measurements

itsmegb, 2023-03-17:
**Describe the bug**
Since the most recent update, openScale has stopped recording measurements from my scale. It was working without issue prior to the update.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the Scale using the Bluetooth button. Toast popup says 'connected' and Bluetooth icon shows as connected.
3. Step on the scale and take measurement
4. Measurement does not appear in the app.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): Not Tested

**Expected behavior**
When a measurement is taken, it should be recorded in the app.

**Additional context**
Looking at the debug log, it appears that openScale requires the `android.permission.BLUETOOTH_PRIVILEGED` permission.
I've reviewed the permissions for the app, and i have all permissions enabled (Location, Notifications and Nearby Devices.

Phone is Google Pixel 7 Pro running Android 13 with 2023.02.05 Security update.
openScale version: v2.5.1-pro (63)

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
Personal Information removed:
[openScale_2023-03-17_09-57.txt](https://github.com/oliexdev/openScale/files/11000258/openScale_2023-03-17_09-57.txt)

apozas, 2023-04-10:
I am having the same issue on a OnePlus 3T running LineageOS 18.1 (Android 11), build 20221024. The debug log is attached.

[openScale_2023-04-06_07-35.txt](https://github.com/oliexdev/openScale/files/11190019/openScale_2023-04-06_07-35.txt)

nekohayo, 2024-01-14:
I believe that this is just that it's not refreshing the view to show the recorded measurement, see https://github.com/oliexdev/openScale/issues/1018#issuecomment-1890959679

maltekluever, 2024-05-23:
I got the medisana BS 444 and have a Fairphone 4 with /e/OS 1.18. Unfortunately for me the synchronisation of the scale records with the app is working not reliably. Sometimes it updates the current measurement, but most of the times it doesn't. Also downgrading to older versions didn't fix the problem.