oliexdev/openScale #944: Black fonts on grey backgrounds are hard to read

glocalglocal, 2023-03-17:
With dark themes, black fonts are difficult to read on grey backgrounds. In broad daylight, it becomes practically impossible. 

[image]

alpapan, 2023-04-25:
Yes. It's also almost impossible to set the person's bday when you first install the app and use it.

Took me a couple of minutes to even realise there was supposed to be a button and almost gave up.
 
[image]

oliexdev, 2023-07-22:
Yeah I currently changing the Android theme to the new Material 3 but I still have some troubles. Any help are welcome. The first screenshot should be fixed in the dev version the second currently not.

oliexdev, 2023-07-29:
should be fixed.

closed 2023-07-29 by oliexdev