oliexdev/openScale #950: Another OKOK scale

arnelap, 2023-03-29:
**Scale name**
Nameless chinese stuff - okok international
https://play.google.com/store/apps/details?id=com.chipsea.btcontrol.en&hl=en&gl=US

I tried the latest version (v2.5.1 (63)) and clicked the "detect blueooth" button before and after stepping on the scale, but it never seems to recognize it. 

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

reports weight only (i think)
Weight values (oldest to newest):
81.50 kg ( 1FD6 )
50.75 kg ( 13D3 )
84.15 kg ( 20DF )

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex (male/female), body height, age, activity level
male, 180cm, 37, active (but I don't think any of this is relevant with this device)

_--> Attach the 1. HCI Snoop log file here <--_
[hci_snoop20230329140200.cfa.zip](https://github.com/oliexdev/openScale/files/11097992/hci_snoop20230329140200.cfa.zip)

_--> Attach the openScale debug log file here <--_
openscale did not see the scale at all, I used the BLE Scannner, it showed 10FFC05B000017700A01245D0249A63DE8 as "raw data" for the MAC address of the scale

Mac address to search for in Bluetooth snoop: 5d:02:49:a6:3d:e8

I don't think any data is sent from the phone to the scale, the scale just advertises the weight while it's working.

arnelap, 2023-03-29:
here is my  guesstimation of what the hex codes mean (note that I have no experience analyzing bluetooth data and i only have 1 scale to test with)

For example
[81.50 kg ( 1FD6 )] > c0 b1 1f d6 17 70 0a 01 25 5d 02 49 a6 3d e8 ae  

c0 = always the same, device id or "magic start bit"?
b1 = counts up while measuring, stops counting when measurement locked
1f d6 =  weight
17 70 0a 01 = i don't know, doesnt seem to change
25 = final measurement locked (24 = is measuring)
5d 02 49 a6 3d e8  = mac of the scale
ae = signal strength? or maybe a checksum but i couldn't reproduce the  checksum if it is one. 

Thanks!

Florin9doi, 2023-04-08:
https://github.com/Florin9doi/openScale/commit/0e0b90dcdf9b29b71349c65b8dac704cfc044663

Florin9doi, 2023-04-08:
* I own a scale with a similar protocol, branded Myria MY4836.
* First issue is that these devices don't have a Bluetooth name. In openScale the names are used to select the protocol and the devices without a name are ignored.
* Another issue is that manufacturerSpecificData can't be used to identify the device/protocol having random values
* Here is my first attempt to support this protocol (and probably the last): 

https://github.com/Florin9doi/openScale/commit/0e0b90dcdf9b29b71349c65b8dac704cfc044663

arnelap, 2023-04-13:
Great work @Florin9doi ! Is there is anything I can do to help test or debug etc don't hesitate to let me know.

Florin9doi, 2024-10-19:
PR: #1081

PR #1081 (2024-11-17): Support for nameless OKOK scales (Myria MY4836)
Closes #950