oliexdev/openScale #954: Beurer BF700 connected but no data sent

iSnackyCracky, 2023-04-17:
**Describe the bug**
Didn't use the scale for a while and it seems the batteries had died. So today I replaced them and tried to connect the App.
The app shows "connected" (and the scale also shows its bluetooth icon) but no data is transmitted. Neither history measurements, nor a current weight when stepping onto the scale while being connected.

Tried to disconnect and re-pair the bluetooth device. "Disconnected the scale in my phones bluetooth-settings but the issue is the same. Issue stays the same. I don't even get a "pairing" screen where I could enter the PIN and the scale doesn't turn on either.
(When trying to pair using the Android settings, the scale immediately turns on and shows a PIN.)

I've also tried deleting the app data, uninstall/reinstall the app, reset the scale multiple times, took out batteries again, restarted phone multiple times.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): openScale v2.5.1-pro

**Debug log**
```
2023-04-17 11:06:32.532 Debug [2] AboutPreferences: Debug log enabled, openScale v2.5.1-pro (63), SDK 33, OnePlus LE2123
2023-04-17 11:06:32.544 Debug [2] AboutPreferences: Selected user id(1) name(snacky) birthday(**********************************) age(**) body height(******) scale unit(kg) gender(male) initial weight(*****) goal enabled(true) goal weight(*****) goal date(Tue Oct 17 00:00:00 GMT+02:00 2023) measure unt(cm) activity level(0) assisted weighing(false)
2023-04-17 11:06:36.509 Debug [2] OpenScale: Disconnecting from bluetooth device
2023-04-17 11:06:36.510 Debug [2] BluetoothCommunication: Bluetooth disconnect
2023-04-17 11:06:36.510 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-04-17 11:06:36.510 Info [2] BluetoothPeripheral: peripheral '5C:CA:D3:00:DC:C8' is disconnecting
2023-04-17 11:06:36.525 Debug [2] MainActivity: Bluetooth connection successful disconnected
2023-04-17 11:06:36.530 Info [2] BluetoothPeripheral: force disconnect 'BF700' (5C:CA:D3:00:DC:C8)
2023-04-17 11:06:36.573 Info [311] BluetoothPeripheral: disconnected 'BF700' on request
2023-04-17 11:06:36.579 Debug [2] BluetoothCommunication: disconnected 'BF700' with status 0
2023-04-17 11:06:40.036 Debug [2] MainActivity: Main Activity Bluetooth permission check
2023-04-17 11:06:40.039 Debug [2] MainActivity: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2023-04-17 11:06:40.039 Debug [2] MainActivity: Permission already granted: android.permission.BLUETOOTH_SCAN
2023-04-17 11:06:40.040 Debug [2] MainActivity: Permission already granted: android.permission.BLUETOOTH_CONNECT
2023-04-17 11:06:40.047 Debug [2] OpenScale: Trying to connect to bluetooth device [5C:CA:D3:00:DC:C8] (BF700)
2023-04-17 11:06:40.050 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2023-04-17 11:06:40.059 Info [2] BluetoothCentralManager: scan started
2023-04-17 11:06:40.060 Debug [2] BluetoothCommunication: Stop machine state
2023-04-17 11:06:40.190 Debug [2] BluetoothCommunication: Found peripheral 'BF700'
2023-04-17 11:06:40.196 Info [2] BluetoothCentralManager: scan stopped
2023-04-17 11:06:41.198 Debug [2] BluetoothCommunication: Try to connect to BLE device 5C:CA:D3:00:DC:C8
2023-04-17 11:06:41.310 Info [2] BluetoothPeripheral: connect to 'BF700' (5C:CA:D3:00:DC:C8) using transport LE
2023-04-17 11:06:41.332 Info [2] BluetoothPeripheral: peripheral '5C:CA:D3:00:DC:C8' is connecting
2023-04-17 11:06:42.899 Info [286] BluetoothPeripheral: connected to 'BF700' (NONE) in 1,6s
2023-04-17 11:06:42.906 Debug [2] BluetoothPeripheral: discovering services of 'BF700' with delay of 0 ms
2023-04-17 11:06:42.914 Info [296] BluetoothPeripheral: discovered 5 services for 'BF700'
2023-04-17 11:06:42.916 Debug [2] BluetoothCommunication: connected to 'BF700'
2023-04-17 11:06:42.917 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2023-04-17 11:06:42.917 Debug [2] BluetoothCommunication: Resume machine state
2023-04-17 11:06:42.918 Debug [2] BluetoothCommunication: Step Nr 0
2023-04-17 11:06:42.918 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2023-04-17 11:06:42.919 Debug [2] BluetoothCommunication: Stop machine state
2023-04-17 11:06:42.929 Debug [2] MainActivity: Bluetooth connection successful established
2023-04-17 11:06:46.914 Debug [286] BluetoothPeripheral: connection parameters: interval=50.0ms latency=0 timeout=6s
2023-04-17 11:06:55.821 Debug [2] OpenScale: Disconnecting from bluetooth device
2023-04-17 11:06:55.822 Debug [2] BluetoothCommunication: Bluetooth disconnect
2023-04-17 11:06:55.822 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-04-17 11:06:55.822 Info [2] BluetoothPeripheral: peripheral '5C:CA:D3:00:DC:C8' is disconnecting
2023-04-17 11:06:55.835 Debug [2] MainActivity: Bluetooth connection successful disconnected
2023-04-17 11:06:55.839 Info [2] BluetoothPeripheral: force disconnect 'BF700' (5C:CA:D3:00:DC:C8)
2023-04-17 11:06:55.850 Info [311] BluetoothPeripheral: disconnected 'BF700' on request
2023-04-17 11:06:55.856 Debug [2] BluetoothCommunication: disconnected 'BF700' with status 0

```

githubbapoopa, 2023-05-08:
Did you try to connect via openscale? First remove the scale in the Andoid Bluetooth settings then connect from inside of openscale. That way is asks for the PIN and after you entered it the scale will connect. Awkward yes, but works for me.

iSnackyCracky, 2023-05-11:
> Did you try to connect via openscale? First remove the scale in the Andoid Bluetooth settings then connect from inside of openscale. That way is asks for the PIN and after you entered it the scale will connect. Awkward yes, but works for me.

As stated that didn't work: "Tried to disconnect an re-pair the bluetooth device. Disconnected the scale in my phones bluetooth-settings but the issue stays the same. I don't even get a "pairing" screen where I could enter the PIN and the scale doesn't turn on either.
(When trying to pair using the Android settings, the scale immediately turns on and shows a PIN.)"

The App just says "connected" when turning on Bluetooth, but nothing happens - scale does not turn on, does not show a Bluetooth pairing PIN, does not transmit any weight-data.

However I just tried the GitHub ("non-pro") version again (completely uninstalled and installed the app again) and somehow it works again.
Now it even is functional in the pro version again as well (re-imported a backup from the non-pro version).

Not really sure what happened in the meantime though..

closed 2023-05-11 by iSnackyCracky