oliexdev/openScale #957: Yunmai M1501 does not retrieve historical data

ValdikSS, 2023-05-26:
**Describe the bug**
openScale is unable to retrieve historical data from Yunmai M1501 scale, while the official Yunmai application supports this function (double-checked with the tests).
Moreover, openScale has retrieved the historical data once, but now I'm unable to reproduce that.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Yunmai M1501 scale in openScale
2. Weight yourself with Bluetooth enabled once, to check that it works
3. Disable Bluetooth connection, weight yourself once more
4. Enable Bluetooth, wait until openScale connects to the device — no historical data

Version v2.5.1 (63)

**Expected behavior**
openScale retrieves historical data from the scale

**Additional context**
I assume that the older protocol is implemented in openScale, made for YunmaiSE_Mini (according to the debug log). The debug log shows that the weight information transferred only upon completion, however official application shows the weight in real time. That's why it's probably an improved protocol.

**Debug log** (slightly edited)
[openScale_2023-05-26_17-54.zip](https://github.com/oliexdev/openScale/files/11576511/openScale_2023-05-26_17-54.zip)

ValdikSS, 2023-05-26:
There's also a difference in the interface when using openScale compared to the official application.

When I weight with bluetooth disabled in openScale, the top line is blinking upon completion, however it does not do that with official application.

During my tests just now, when the scale is first used with the official application and then connected to openScale, openScale does retrieve historical data. This is probably the protocol.

ValdikSS, 2023-05-27:
According to [reverse-engineered protocol reference](https://gist.github.com/conoro/f0c1d96c450a8f5cce70e2846c3686c4), openScale implements 1e protocol version, while my scale uses 1f.

Quote from the document:
```
Example command: 0d 05 13 00 16
Example response: 0d 1e 05 05 1e
```

However, my scale responds with `0d 1F 05 05 1F`

ValdikSS, 2023-05-27:
@oliexdev, for the historical data to be retrieved, the "sync time" command must contain 0x14 in the 8th byte.

I want to make it 0x14 only if the protocol version is 1F, but it seems I can't receive any data inside `onBluetoothNotify()` until the state machine processes steps. I tried to call `setNotificationOn()` as the first step, but that did not help.
Please tell me how can I receive the notification data inside state machine.

oliexdev, 2023-07-22:
The command  `writeBytes(WEIGHT_CMD_SERVICE, WEIGHT_CMD_CHARACTERISTIC, magic_bytes); ` has to be send to retrieve something inside `onBluetoothNotify()`

Then you should retrieve your respond.

ValdikSS, 2023-07-24:
I've played with the code over several days in May, but I still could not make the scale to send historical data reliably. When something is not right with the configuration, the "progress bar" on the scale blinks after weight process, and either this data is not saved into internal memory or something other happens.

Don't know the reason yet. The only thing this is really different from the official application and openScale is that openScale uses modify-with-create opcode upon adding new person while the official almost always prefers add-new opcode. But this is also not reliable and openScale worked fine with my modifications over the course of several days, and then it stopped working again.