oliexdev/openScale #962: Bug when importing muscle measurements

danicosas, 2023-06-04:
**Describe the bug**
Having configured in the app that the muscle measurements appear in kilograms, when I import a correct .csv file with these measurements, the app imports them incorrectly.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the **side menu** >> **Settings** >> **Measurements** >> **Muscle** config icon
2. Make sure **Measurement in %** is **off**
3. Go to **Overview** section >> **Top right menu** with 3 points >> **Import**
4. We import a correct .csv >> [2023_cleandata_bug.csv](https://github.com/oliexdev/openScale/files/11645114/2023_cleandata_bug.csv)
5. We observe that the measurements of the **Muscle value** do not match with the .csv

Reproduced with [latest release version](https://github.com/oliexdev/openScale/releases/tag/v2.5.1): v2.5.1

**Expected behavior**
That the measurements in kilograms are imported as they are in the .csv

**Debug log**
[openScale_2023-06-04_09-36.txt](https://github.com/oliexdev/openScale/files/11645102/openScale_2023-06-04_09-36.txt)

**Attachments**
[image]
[image]
[image]

oliexdev, 2023-07-22:
The base unit of the muscle in CSV is always in %. The values in the CSV are in %. If you imported it with the option "Measurement in %" off you see the value in your weight unit.

closed 2023-07-22 by oliexdev

danicosas, 2023-07-25:
I already did it as you say, before opening the bug, and the error is reproduced. But if it doesn't work for you to fix it, I'll fork it and fix it myself. Otherwise, great job.

oliexdev, 2023-07-29:
Again its not a bug but what is your purpose? If you export it with openScale and import it again everything works. IF you want import a different CSV files which use kg in the CSV file then convert it in the CSV files before you import it in openScale.

danicosas, 2023-07-29:
Indeed, the export from Xiaomi comes as it comes. In the bug I attached an excel file exported from my official Xiaomi account. (Obviously for privacy, I've only left some real logs for the bug.)
It does not make sense that in a data import functionality, you have to modify the data from the official source.
But again, if it's too much trouble to fix, I'll fork and fix it myself.

oliexdev, 2023-07-29:
Yes it would too much trouble to implement each individual 3rd party CSV version into openScale. Therefore we have on the wiki useful import scripts https://github.com/oliexdev/openScale/wiki/Useful-import-scripts ;)