oliexdev/openScale #973: Ryot sync support

IgnisDa, 2023-07-26:
Hello, awesome project! I am the developer of [Ryot](https://github.com/IgnisDa/ryot). I have started working on add fitness tracking support to my project and was recommended to check this project out.

I see that openScale has support for wger. Would you be willing to accept contributions for syncing with Ryot directly?

It would take me a couple of months to complete the fitness tracking support for Ryot and would like to add sync support after that.

oliexdev, 2023-07-29:
Sure but I could implement that in openScale sync if it has a easy to use API and not directly in openScale.

IgnisDa, 2023-07-29:
Yes, that is what I meant.

> easy to use API

What kind of API is needed?

oliexdev, 2023-07-30:
How are you typically implement your API? wger use the following REST API https://wger.de/api/v2/

oliexdev, 2023-07-30:
But what I normally need for syncing are at least the following methods:

```
insert new measurement
delete a measurement
update a  measurement
get a list of all stored measurements
clear all measurements
```

IgnisDa, 2023-07-30:
Cool, will update here when I implement them.

IgnisDa, 2023-08-05:
@oliexdev I added graphql queries/mutations for adding measurements.

First login to https://ryot.fly.dev and then https://ryot.fly.dev/settings/tokens generate a token. All requests to the graphql endpoint must have the header: `X-Auth-Token`: `<copied-token>`.

You can inspect the request and response structures in https://ryot.fly.dev/graphql.

- insert: `mutation createUserMeasurement`
- delete: `mutation deleteUserMeasurement`
- update: delete and create
- list: `query userMeasurementsList`
- clear: list, and delete in a loop

Please ping me here if anything else is needed.

IgnisDa, 2023-08-19:
Hello @oliexdev, any updates on this? Anything I can do to help?

oliexdev, 2023-08-19:
I am not familiar with the graphQL, it seems more complex than the wger REST API. I am using [retrofit2](https://github.com/square/retrofit) to interact with the wger REST API.

Are you familiar with Android programming in Java? Or could you provide some curl examples for each operation?

IgnisDa, 2023-08-23:
Yes, I am familiar. A graphql request can be converted to curl easily. Here is the request for creating a measurement.

```bash
curl 'https://ryot.fly.dev/graphql' --compressed -X POST -H 'Content-Type: application/json' -H 'Referer: https://ryot.fly.dev/fitness/measurements' --data-raw '{"query":"mutation CreateUserMeasurement($input: UserMeasurementInput!) {\n  createUserMeasurement(input: $input)\n}","variables":{"input":{"timestamp":"2023-08-23T08:08:28.676Z","name":"This is a test.","stats":{"weight":"78.900","muscle":"34.560"},"comment":"Some comment."}},"operationName":"CreateUserMeasurement"}'
```

You can do it yourself by opening the network tab and using the "Copy as Curl" button for any graphql endpoint.

alexanderadam, 2024-04-20:
How come that this feature request has the label `question`?

IgnisDa, 2024-04-21:
For anyone following this, Ryot can now import from Openscale (`v5.0.4`).

alexanderadam, 2024-04-21:
> For anyone following this, Ryot can now import from Openscale (`v5.0.4`).

but this is a _manual_ sync then, right?

And you probably created this issue back then to have an automatic sync via [openScale sync](https://github.com/oliexdev/openScale/wiki/openScale-sync), is that correct?

Either way I'd love to see an _automatic_ sync. :wink:

IgnisDa, 2024-04-21:
> but this is a manual sync then, right?

Yes, it is meant as a one time import.

> And you probably created this issue back then to have an automatic sync via [openScale sync](https://github.com/oliexdev/openScale/wiki/openScale-sync), is that correct?

Yes, I want Ryot sync to be added to Openscale Sync too hence this issue is still open.

GitHubUser325, 2025-05-23:
+1 for Ryot Sync Support!

Additionally:
When exporting to a csv, the values BMI, BMR, TDEE are not transferred. Can these please be included in the exported csv? Thank you for your help :)

sodalimes, 2025-06-07:
+1 for ryot support

LostName1, 2025-06-11:
+1 this would be great