oliexdev/openScale #987: Android App crashes when Changing the Name

ThomasLagemann, 2023-08-26:
In an account with 2 users, when assigning a measurement to the other user the app crashes an must be restarted.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up two users
2. Go to the list of measurements
3. Click edit
4. In the Name dropdown select the other user
5. Click save
6. App crashes

 

Reproduced with:
Build version: 2.5.2-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2023-08-26 09:24:52 
Device: Google Pixel 6a 
OS version: Android 13 (SDK 33) 

**Expected behavior**
The measurement should be assigned to the other user. App should not crash.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
Build version: 2.5.2-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2023-08-26 09:24:52 
Device: Google Pixel 6a 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: scaleMeasurements.userId, scaleMeasurements.datetime (code 2067 SQLITE_CONSTRAINT_UNIQUE)
	at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
	at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(SQLiteConnection.java:913)
	at android.database.sqlite.SQLiteSession.executeForChangedRowCount(SQLiteSession.java:756)
	at android.database.sqlite.SQLiteStatement.executeUpdateDelete(SQLiteStatement.java:67)
	at androidx.sqlite.db.framework.FrameworkSQLiteStatement.executeUpdateDelete(FrameworkSQLiteStatement.kt:38)
	at androidx.room.EntityDeletionOrUpdateAdapter.handle(EntityDeletionOrUpdateAdapter.kt:62)
	at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.update(ScaleMeasurementDAO_Impl.java:172)
	at com.health.openscale.core.OpenScale.updateScaleMeasurement(OpenScale.java:416)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.saveScaleData(MeasurementEntryFragment.java:346)
	at com.health.openscale.gui.measurement.MeasurementEntryFragment.onOptionsItemSelected(MeasurementEntryFragment.java:200)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3284)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:3168)
	at androidx.fragment.app.Fragment.performOptionsItemSelected(Fragment.java:3288)
	at androidx.fragment.app.FragmentManager.dispatchOptionsItemSelected(FragmentManager.java:3168)
	at androidx.fragment.app.FragmentManager$2.onMenuItemSelected(FragmentManager.java:503)
	at androidx.core.view.MenuHostHelper.onMenuItemSelected(MenuHostHelper.java:107)
	at androidx.activity.ComponentActivity.onMenuItemSelected(ComponentActivity.java:557)
	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:264)
	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:269)
	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:110)
	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:66)
	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:225)
	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:781)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:977)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:156)
	at android.view.View.performClick(View.java:7542)
	at android.view.View.performClickInternal(View.java:7519)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:29476)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7918)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)

oliexdev, 2024-04-13:
you have a measurement with the exact same date/time stamp, that's not allowed. please try to change the time/date slightly before you assign the other user.

closed 2024-04-13 by oliexdev

ThomasLagemann, 2024-04-13:
Finding and deleting entries with duplicated timestamps is a workaround not a fix. A fix would be to prevent the app from creating problemstic entries by checking if a timestamp already exists.