oliexdev/openScale #991: When connecting to Renpho ‎ES-CS20M-RD app chrashews.

kambancode, 2023-09-10:
**Describe the bug**
When connecting to Renpho ‎ES-CS20M-RD app chrashews.

**To Reproduce**
Steps to reproduce the behavior:
1. open app
2. start scale
3. app reports connectiong to scale
4. app crashes

**Debug log**
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2023-09-10 11:51:57 
Device: HUAWEI CLT-L29 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
 at j$.util.Objects.requireNonNull(Unknown Source:5)
 at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
 at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
 at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
 at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
 at android.os.Handler.handleCallback(Handler.java:900)
 at android.os.Handler.dispatchMessage(Handler.java:103)
 at android.os.Looper.loop(Looper.java:219)
 at android.app.ActivityThread.main(ActivityThread.java:8668)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)

iamstevedavis, 2023-09-13:
Adding my logs as well.
Scale: Renpho ES-CS20M

```
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2023-09-13 17:41:01 
Device: Google Pixel 6a 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7918)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
3m
```

floriansto, 2023-10-07:
I can confirm this, too.

kiers, 2023-10-18:
I too have ES-CS20M from amazon.  The app crashes on bluetooth approach to scale. I have allowed Bluetooth permission in app settings.
Here is my log

Build version: [2.5.2](tel:2.5.2)
Build date: [1981-01-01 01](tel:1981-01-01 01):01:02
Current date: [2023-10-17 22](tel:2023-10-17 22):46:44
Device: Samsung SM-G[991](tel:991)U1
OS version: Android 12 (SDK 31)

Stack trace: 
[java.lang.NullPointerException](http://java.lang.nullpointerexception/): no valid characteristic provided
at j$.[util.Objects.requireNonNull](http://util.objects.requirenonnull/)(Unknown Source:5)
at [com.welie.blessed.BluetoothPeripheral.writeCharacteristic](http://com.welie.blessed.bluetoothperipheral.writecharacteristic/)([BluetoothPeripheral.java:1229](http://bluetoothperipheral.java:1229/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes](http://com.health.openscale.core.bluetooth.bluetoothcommunication.writebytes/)([BluetoothCommunication.java:303](http://bluetoothcommunication.java:303/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes](http://com.health.openscale.core.bluetooth.bluetoothcommunication.writebytes/)([BluetoothCommunication.java:291](http://bluetoothcommunication.java:291/))
at [com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep](http://com.health.openscale.core.bluetooth.bluetoothqnscale.onnextstep/)([BluetoothQNScale.java:111](http://bluetoothqnscale.java:111/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep](http://com.health.openscale.core.bluetooth.bluetoothcommunication.nextmachinestep/)([BluetoothCommunication.java:610](http://bluetoothcommunication.java:610/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep](http://com.health.openscale.core.bluetooth.bluetoothcommunication.nextmachinestep/)([BluetoothCommunication.java:612](http://bluetoothcommunication.java:612/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep](http://com.health.openscale.core.bluetooth.bluetoothcommunication.nextmachinestep/)([BluetoothCommunication.java:612](http://bluetoothcommunication.java:612/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState](http://com.health.openscale.core.bluetooth.bluetoothcommunication.resumemachinestate/)([BluetoothCommunication.java:210](http://bluetoothcommunication.java:210/))
at [com.health.openscale.core.bluetooth.BluetoothCommunication](http://com.health.openscale.core.bluetooth.bluetoothcommunication/)$[1.onServicesDiscovered](http://1.onservicesdiscovered/)([BluetoothCommunication.java:464](http://bluetoothcommunication.java:464/))
at [com.welie.blessed.BluetoothPeripheral](http://com.welie.blessed.bluetoothperipheral/)$1$[1.run](http://1.run/)([BluetoothPeripheral.java:210](http://bluetoothperipheral.java:210/))
at [android.os.Handler.handleCallback](http://android.os.handler.handlecallback/)([Handler.java:938](http://handler.java:938/))
at [android.os.Handler.dispatchMessage](http://android.os.handler.dispatchmessage/)([Handler.java:99](http://handler.java:99/))
at [android.os.Looper.loopOnce](http://android.os.looper.looponce/)([Looper.java:226](http://looper.java:226/))
at [android.os.Looper.loop](http://android.os.looper.loop/)([Looper.java:313](http://looper.java:313/))
at [android.app.ActivityThread.main](http://android.app.activitythread.main/)([ActivityThread.java:8751](http://activitythread.java:8751/))
at [java.lang.reflect.Method.invoke](http://java.lang.reflect.method.invoke/)(Native Method)
at [com.android.internal.os.RuntimeInit](http://com.android.internal.os.runtimeinit/)$[MethodAndArgsCaller.run](http://methodandargscaller.run/)([RuntimeInit.java:571](http://runtimeinit.java:571/))
at [com.android.internal.os.ZygoteInit.main](http://com.android.internal.os.zygoteinit.main/)([ZygoteInit.java:1135](http://zygoteinit.java:1135/))

iff7378, 2023-12-11:
Same scale, same problem:

Why is it still listed as compatible in the wiki??

Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2023-12-11 18:09:37 
Device: Google Pixel 6 Pro 
OS version: Android 14 (SDK 34) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8194)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)

iff7378, 2023-12-17:
The SDK for this scale is here (and all scales that show up as QN Scale): https://github.com/YolandaQingniu/qnscalesdkX

I wish I had the dev skills to help implement this but I'm not a software person. I'd really like to see this fixed.

@oliexdev If there's other ways I can help get this issues fixed, I'm all ears.

antoinebou12, 2024-03-08:
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2024-03-08 15:58:55 
Device: Google Pixel 6a 
OS version: Android 14 (SDK 34) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:959)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8501)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)

rfdrake, 2024-07-28:
I believe this is where the problem is (from https://yolandaqingniu.github.io/en/flow/ble_scale_custom_connect.html)

```
Our scale uses two sets of services, and one device will only contain one set. The APP needs to traverse all services to find one set of service IDs. If one set is found, the feature values that need to be called back are enabled according to the feature values in the following table.
```

The driver for QN-Scale uses FFE0.  The other service ID is FFF0.   The characteristic UUIDs are different.

Here is the mentioned table for characteristics of FFF0:
```
FFF1 	Notify 	Data callback feature value
FFF2 	Write 	Data write characteristic value
2A19 	Read 	Charging scale electric quantity characteristic value
```

So the question would be how to query the scale for the service ID it wants.  And if there would be a place to store it so that it doesn't need to probe on every connection.  Finally, I wonder if it would be better to split into two different drivers because the changes might be extensive and confusing. 

This is all theory right now.  I have not yet recompiled with the new values to see if it fixes the crash on my scale.  I'm not even sure how the new values would work because they're missing several parts of the state machine.   I suspect they may have simplified the scale to use less round trips, but sniffing the official apps bluetooth connection may be the only way to really know how to deal with it.

I haven't jumped through the hoops to try the official app yet, but if I do I'll post the snoop log.

rfdrake, 2024-08-05:
I recompiled and tried it using the other service ID and characteristic and it no longer crashes.  It doesn't log the weight either, but I suspect that is because I'm not doing the right thing with the results.  I'll need to run a packet capture to get a better idea of what to do next.

ndurchx, 2024-10-10:
I bought this scale, because it is listed as supported. Please remove it from Wiki.

EDharmasiri, 2024-11-06:
I've put in a PR to support the 2nd set scales (Using a Renpho R-A019). Would appreciate it if any of you can try it out and see if it works for you. Would also appreciate it if anyone with a working 1st set scale can test it out to see if it works without impacting them.

https://github.com/oliexdev/openScale/pull/1083

ndurchx, 2024-11-07:
The Renpho ‎ES-CS20M works with the PR

closed 2025-03-15 by oliexdev

kiers, 2025-03-15:
Terribly sorry but in english please...... there's no version update? The Renpho ES c20 is still listed in the wiki with checkmark (yes) for compatibility in initialisation and body metrics.  ? Is it working you say?

kiers, 2025-03-15:
Ah, there is SOMe progress.
I tried version 2.5.3 (released a few months ago), and:
1) the app can launch
2) it can connect to the scale
3) it shows the choices: table, time graph, history... though they are blank;
4) Then it crashes. It never collects the data from the scale!

Err Log paste:
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-03-15 12:02:05 
Device: Samsung SM-G991U1 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8757)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)

tsteven4, 2025-03-15:
@kiers, the fixes are in the dev build, https://github.com/oliexdev/openScale/releases/tag/dev-build , not 2.5.3.