oliexdev/openScale #994: entering date of birth of a new user does not work well

FransM, 2023-09-29:
**Describe the bug**
I freshly installed openscale and wanted to add a user.
Entering the date of birth seemed impossible. See below for what I did.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to add new user
2. try to modify the date
3. this gives a calendar widget on which I selected a date
4. This date is shown on the left of the calendar
5. It is totally unclear how to get back to the new user form with the new date; pressing back did not work (it kept the default date being the day of entry 20 years ago). There is also a small pencil at the left, that takes over the date but still I cannot get back to the form without loosing the date

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 
v2.5.2(64) (F-droid version as of sep 29, 2023)

**Expected behavior**
I expected one of the following:
- That on clicking on a date I would go back to the user form with the new date selected (imho preferred solution)
- Or that there are cancel/ok buttons to reject/accept the new value

**Additional context**
Hardware: samsung A10 (2016) running LineageOS 19 (Android 12)

Note that at some point (with select all and copy) I was able to copy the date, but I really have no clue how I did this.

FransM, 2023-09-29:
Updating above PR
Afterwards I installed the development version on my Pixel 6A 
For that I did get the cancel/ok button on the date picker.
No idea why this does not show on the Samsung A10 tablet.

FransM, 2023-09-30:
I found the cause, my tablet was in "dark theme" giving the date picker a very dark background on which the buttons cancel/ok were only visible when the brightness is high.

I was about to close this ticket but then I realized I did not have this issue in other apps or in the control panel.
When I went to the control panel and want to change the time the ok and cancel button have a turquoise-like color.

Speculation: Maybe the issue is caused because the app does use black to render the texts instead of some system color.
So I reopened the ticket so someone can verify if there is something that can be done here.

(I"ll see if I can find something myself but I am not really a java wizard and not familiar with the program at all)

FransM, 2023-10-04:
I have been browsing the code and it seems related to theming.

In themes.xml there is this line
    <style name="AppTheme" parent="Theme.Material3.Light.NoActionBar">

I suspect that instead of Light we need DayNight here, but I am not sure what needs to be done in the theme definition.

Alternatively we could create a separate theme and add a .setTheme when building the datepicker in UserSettingsFragment.java.
This call
                MaterialDatePicker<Long> datePicker = MaterialDatePicker.Builder
                        .datePicker()
                        .setSelection(birthday.getTime())
                        .build();

But maybe there are other places where this is an issue, so I suspect that changing it in themes.xml is the better solution

oliexdev, 2024-11-17:
should be fixed now with the latest commit https://github.com/oliexdev/openScale/commit/8b4326687c91d33f496a0f37dc5fc7e14388aca9

closed 2024-11-17 by oliexdev