oliexdev/openScale #998: MQTT: Enhance Publishing in openScaleSync to Include All Metrics

Sikerdebaard, 2023-10-11:
**Feature Request:**

I've observed that the openScaleSync app currently has support for publishing data to an MQTT broker, which is fantastic. However, it seems that only the date and weight metrics are being sent to the broker.

**Desired Enhancement:**

To improve the utility and flexibility of this feature, I propose expanding the MQTT publishing functionality to include all available metrics from the openScaleSync app. This will provide users with a comprehensive dataset on the broker, allowing for richer data analysis and integration possibilities.

**Benefits:**

-     Enables users to capture a complete health profile via MQTT, enhancing data-driven insights.
-     Increases interoperability with other systems and tools that might utilize these additional metrics.
-     Enhances user satisfaction by providing more granular control over data sharing.

**Potential Challenges/Considerations:**

While sending more data might increase network traffic slightly, I believe the added benefits outweigh the potential drawbacks. Nevertheless, consider allowing users to choose which metrics they'd like to publish to offer flexibility and control over the amount of data being sent.

**Offer to Contribute:**

I'm keen on contributing to this enhancement in the openScaleSync app. However, I encountered a stumbling block: the source for openScaleSync seems to be unavailable, and I couldn't locate it. If it's possible to provide access or guide me to the repository, I'd be more than willing to assist in code development and implementation of this feature.

NZX-DeSiGN, 2023-10-11:
I'm in the same situation; I would like to retrieve the body fat percentage in Home Assistant. 

Syncing to Google Fit doesn't transfer this data, so I thought I'd fill the gap with MQTT, but unfortunately no 🙁

jmdurant, 2024-02-24:
I would love to see this as well.  It could open the door for so many uses of the data. Thanks!

LoghamLogan, 2024-03-02:
I'd also like to see this!

oliexdev, 2025-06-25:
please try out the latest openScale sync version https://play.google.com/store/apps/details?id=com.health.openscale.sync.oss you can also contribute to the soure code at https://github.com/oliexdev/openScale-sync

closed 2025-06-25 by oliexdev