oliexdev/openScale #1001: measurement assigned to wrong user

FransM, 2023-10-12:
**Describe the bug**
In the scenario as described in https://github.com/oliexdev/openScale/issues/1000 the measurement was assigned to me

**To Reproduce**
Could not reproduce

**Expected behavior**
My measurement is about 100 kg. The new measurement was about 15 kg. Still this was added to my list.
I would expect that if the weight changes so much that this is treated as error or is used as a popup to ask for the user.
(or maybe add a user unknown or unassigned)

**Additional context**
Xiaomi body composition scale 2

**Debug log**
not available, cannot reproduce and only noticed afterwards.

FransM, 2023-10-19:
I've peeked into the code and I suspect that I know what happened.
Smart user assignment was on but the newly added measurement was out of range for any user so the measurement was assigned to the selected user as ignore-out-of-range was off. This is probably as intended but in my case subsequent measurements ended up at the wrong user.

closed 2023-10-19 by FransM

Strubbl, 2024-12-20:
In this case, do you know if it is possible to move the measurement to the correct user anyhow?

Strubbl, 2024-12-20:
Found the answer in comment https://github.com/oliexdev/openScale/issues/999#issuecomment-2053566174