oliexdev/openScale #1002: Unable to import data again after importing once before even after deleting.

Savasdotexe, 2023-10-13:
**Describe the bug**
Wanted to merge data logs from external/manual readings previously done so I exported then added the logs manually csv file, but noticed that some were not appearing.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to top right menu (...), export csv file. You should have some user data/readings.
2. Delete all logs/readings manually by going to Table > select reading, bin icon.
3. Go back and Import same file (no need to edit)
4. See error (notice no data exported), or repeat process from step 2 to reproduce.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): ???

**Expected behavior**
Should be exporting fully the entire list even after deleting.

**Additional context**
I was able to resolve by deleting app data/clearing cache.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

oliexdev, 2024-04-13:
that's because openScale will not really delete the measurement it will only markes as deleted because some scales will always transfer ALL measurements.

closed 2024-04-13 by oliexdev

Savasdotexe, 2024-04-13:
The problem is after trying to import, they don't appear, so I guess, as you say, because it still thinks it's there.