oliexdev/openScale #1003: assign out-of-range measurements to a dummy pseudo user (e.g. other)

FransM, 2023-10-19:
**Is your feature request related to a problem? Please describe.**
context: smart user assignment is on, ignore out of range is off.
Consider two users A and B.
User A weights 80 kg, user B weights 75 kg.
User A is the active user.
Now a kid weighting 25 kg steps onto the scale. There is no user associated with the kid.
The measurement gets assigned to the active user (selectedScaleUser)
After that smart assignment totally screws up as following weightings of user A will be assigned to B (as the last weight of A is now 25 kg).

**Describe the solution you'd like**
If the measurement is out of range for all users assign it to a pseudo user named "other" or "unassigned" or "unclassified" or so.

**Describe alternatives you've considered**
An alternative could be to come with a popup asking the user to which user to assign the measurement (or still discard it)

FransM, 2023-10-19:
related to this: if you are assigned an out-of-range measurement subsequent smart user assignments could either ignore the out-of-range measurement, taking the last one that is not out of range, or look at the history (so e.g. look at the median or average weight (maybe even weighted over time)

oliexdev, 2024-04-13:
> related to this: if you are assigned an out-of-range measurement subsequent smart user assignments could either ignore the out-of-range measurement, taking the last one that is not out of range, or look at the history (so e.g. look at the median or average weight (maybe even weighted over time)

just enable in the settings the option ignore out of range data.

closed 2024-04-13 by oliexdev