oliexdev/openScale #1006: SyncService crashing with Starting FGS without a type callerApp=ProcessRecord{6ba7a5c 13848:com.health.openscale.sync/u0a301} targetSDK=34

liutikas, 2023-10-23:
**Describe the bug**
SyncService is crashing in the background

```
java.lang.RuntimeException: Unable to create service com.health.openscale.sync.core.service.SyncService: android.app.MissingForegroundServiceTypeException: Starting FGS without a type  callerApp=ProcessRecord{1da4036 13929:com.health.openscale.sync/u0a301} targetSDK=34

android.app.MissingForegroundServiceTypeException
at com.health.openscale.sync.core.service.SyncService.onCreate
  at android.app.ActivityThread.handleCreateService(ActivityThread.java:4664)
  at android.app.ActivityThread.-$$Nest$mhandleCreateService(Unknown Source:0)
  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2264)
  at android.os.Handler.dispatchMessage(Handler.java:106)
  at android.os.Looper.loopOnce(Looper.java:205)
  at android.os.Looper.loop(Looper.java:294)
  at android.app.ActivityThread.main(ActivityThread.java:8177)
  at java.lang.reflect.Method.invoke(Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
Caused by: android.app.MissingForegroundServiceTypeException: Starting FGS without a type  callerApp=ProcessRecord{1da4036 13929:com.health.openscale.sync/u0a301} targetSDK=34
  at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:53)
  at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:49)
  at android.os.Parcel.readParcelableInternal(Parcel.java:4870)
  at android.os.Parcel.readParcelable(Parcel.java:4852)
  at android.os.Parcel.createExceptionOrNull(Parcel.java:3052)
  at android.os.Parcel.createException(Parcel.java:3041)
  at android.os.Parcel.readException(Parcel.java:3024)
  at android.os.Parcel.readException(Parcel.java:2966)
  at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6761)
  at android.app.Service.startForeground(Service.java:775)
  at com.health.openscale.sync.core.service.SyncService.onCreate(:8)
  at android.app.ActivityThread.handleCreateService(ActivityThread.java:4651)
  ... 9 more
```

**To Reproduce**
Steps to reproduce the behavior:
1. Regular app usage

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 
openScale `2.5.2-pro`
openScale sync `0.3.5`

**Expected behavior**
App is compliant with https://developer.android.com/about/versions/14/changes/fgs-types-required

**Additional context**
Pixel 8 running Android 14 `UD1A.230803.041`

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

oliexdev, 2024-04-13:
Thanks for reporting, seems Google changes again something in the library...

liutikas, 2024-07-05:
Any updates here? It would be really awesome to have this fixed.

oliexdev, 2025-05-10:
This problem might be resolved in the upcoming beta version (0.4) of openScale sync, which features a complete rewrite of the application and includes improvements to the synchronization logic.

We are currently running a closed beta test for this new version. If you'd like to help test and see if this issue is fixed, please consider joining the beta program. You can find more details and instructions on how to join in our pinned beta testing invitation issue here: https://github.com/oliexdev/openScale/issues/1124

closed 2025-05-10 by oliexdev