oliexdev/openScale #1010: Documentation completeness How-to

mazafra1, 2023-11-08:
Add **Pre-requisites section** to https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

what type of device you need (mobile vs laptop), etc.

PS: sorry to post a bug ticket for this.

WednesdayElectronics, 2023-11-09:
Both mate.
If you follow the guide, you'll see that its getting you to create a log file of the PHONES interaction with the scale, then you use a computer to extract this file from the phone to be posted here. (the Log file is generally inaccsessable via the phone itself)
Spend an evening reading up and some of the fog will lift....
Hope this helps

mazafra1, 2023-11-11:
[openScale_2023-11-11_18-31.txt](https://github.com/oliexdev/openScale/files/13326133/openScale_2023-11-11_18-31.txt)
Here it is!!
Thanks.

closed 2024-04-13 by oliexdev