oliexdev/openScale #1012: Xiaomi scale 2 unclear which model is supported

ebastler, 2023-12-04:
Hello,

the supported scale list lists a "Xiaomi Mi scale v2", however the associated picture looks like the scale sold as a "Xiaomi Body Composition Scale 2", while the "Xiaomi Mi scale v2" on the Mi homepage looks totally different. Can you clarify which of the two models is actually supported?

https://www.amazon.it/Composition-impedenziometrica-Diagnostica-Bluetooth-composizione/dp/B07VD9F419
https://www.amazon.it/Xiaomi-Smart-Bilancia-Persona-bianco/dp/B07VLN38BL

lbschenkel, 2023-12-06:
I purchased a Smart Scale 2 that is the same one as your Amazon link above, and I can confirm that it works with openScale (and other software like Home Assistant).

My understanding that both scales implement the same Bluetooth protocol, the only difference being that the "body composition" version includes impedance data too, which the other one obviously doesn't as it doesn't have the sensor.

ebastler, 2023-12-06:
> My understanding that both scales implement the same Bluetooth protocol, the only difference being that the "body composition" version includes impedance data too, which the other one obviously doesn't as it doesn't have the sensor.

Thank you for the information! I'll give the body composition scale a try, then. Price is almost the same...

lbschenkel, 2023-12-06:
> Thank you for the information! I'll give the body composition scale a try, then. Price is almost the same...

Keep in mind that this is what I've read, I haven't personally tried the "body composition" model with openScale, only the simpler one.

ebastler, 2023-12-09:
I can confirm, the "Xiaomi Body Composition Scale 2" works as well. All parameters are shown in the app. I never installed the official Xiaomi app to double-check if readings like body fat are correct, but since weight is, I'd assume the rest is as determined (or rather, guessed :D) by the scale too.

closed 2023-12-09 by ebastler

marmpa, 2024-09-12:
> I ordered the "Xiaomi Body Composition Scale S400" which is not supported. In the menu to add the scale it is displayed as Xiaomi Scale S400 378A.

It is not yet supported. I don't have much experience, but I'm trying to see if i can figure out how it works, to add it!

macsimc, 2024-09-20:
> > I ordered the "Xiaomi Body Composition Scale S400" which is not supported. In the menu to add the scale it is displayed as Xiaomi Scale S400 378A.
> 
> It is not yet supported. I don't have much experience, but I'm trying to see if i can figure out how it works, to add it!

I got the scale and made some dumps, i cannot descipher the protocol though ...

[btsnoop_hci.log](https://github.com/user-attachments/files/17077753/btsnoop_hci.log)

[Screenshot_2024-09-19-14-37-48-231_com macdom ble blescanner](https://github.com/user-attachments/assets/04fd1bf1-e1e8-4ee1-bb12-3f8b7ecf616b)

macsimc, 2024-09-22:
The messages containing data seem to be encrypted , i tried to decrypt them with the ble key without success
The flow seems to be 

- write a message 01 00 XX [...] to handle x27 - package length is variable between 12 and 256 bytes,sometimes if there is a longer package the rest starts with 02 00 ...
- receive a notification on handle 0x2a in the form: 00 00 00 02 00 [XX+1] 00 [data .. ] 

XX starts with 00  and increases until there are no more history items.

the response data length is variable also, probably depends if it is a full scale read (with body composition) or simple.
the responses do not have any visible pattern and the payload does not repeat during the next captures

manfye, 2024-11-16:
any update on this?

Karsten-Yan, 2024-11-18:
I also purchased the Xiaomi s400 body composition scale. For me it shows as "Xiaomi Scale s400 8881" in the openscale connection menu. Is there any development activity going for this model? If so any logs or something i could provide for this model?

pciavald, 2025-01-23:
On my side I bought this [Xiaomi Body Composition S400](https://www.amazon.fr/dp/B0D3P8ZP21), it shows as `Xiaomi Scale S400 084B`. @ebastler did you do anything special to make it work or was it another reference?

alfureu, 2025-01-28:
Mine shows `Xiaomi Scale S400 D32E` ... any chance of supporting it?

dbsdbs93, 2025-04-28:
Hello:  Has anyone had any luck with this?  I've spent some time in the last few days deciphering the protocol and... I understand the protocol pretty well, but it's pretty clear that the data exchanges are encrypted, probably with AES-CCM.  Was hoping someone with experience could help me figure out how to get at the key and the nonce that's being used.  I assume that if the implementation is done well, I'm not just going to fish it in plaintext out of the pairing BLE capture.