oliexdev/openScale #1018: Last measurement appearence

Inodeus, 2024-01-13:
Hi

The last current measurement doesn't appear on graphs both home and chart sections. Is it possible to add posibility to observe currently added data on graphics?

Thank you.

P.S.: not sure whether it is feature request or bug report so decided to put it that way.

nekohayo, 2024-01-14:
I've experienced this with version 2.5.2 (or recent versions of the last few years). What happens is that the data does get recorded, but doesn't appear on its own as the table view (and presumably others) does not automatically refresh when new measurements are recorded. It only appears once I touch something on the screen, i.e. do a swipe gesture somewhere on the table (like a fake "pull down to refresh").

It would be more convenient and much less confusing if the view refreshed on its own whenever a new measurement is received.

oliexdev, 2025-09-13:
should be fixed in the beta version

closed 2025-09-13 by oliexdev