oliexdev/openScale #1022: Importing backup fails if containing non activated data points

Olangu, 2024-01-28:
**Describe the bug**
Importing backup fails if containing non activated data points

**To Reproduce**
Steps to reproduce the behavior:
1. Enable caliper
2. Log weight with caliper data
3. Export backup
4. Try import on blank installation
5. Program says imported but nothing is.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 2.5.2 (64)

**Expected behavior**
Import data and activate caliper

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

ksiezak, 2024-03-06:
I can confirm this bug. I added bone mass in precents, and I cannot import backup.

ksiezak, 2024-03-18:
Update:
I have tried latest debug version and it worked without a problem.
Next I downgraded to last F-Droid version (v2.5.2 64), and tried to import db.
First atempt failed but when I activated bone mass and set it to percents it worked! :)
I think that I tried this method before with no success, but maybe I am wrong.

closed 2024-04-13 by oliexdev