oliexdev/openScale #1026: Unclear how to use app - assisted user != user ?

SystemZ, 2024-03-15:
**Describe the bug**
I downloaded latest version of app from fdroid.
I configured everything that setup wizard asked.
I step on scale to turn it on and expect that measurement will be added to app.

That didn't worked so I step on scale and press + to add.
App is showing that I need to add assisted user but it doesn't explain how to do that. 
I also already created user and it's selected as default (?) in settings.

Same scale works OK with similar app on iOS.

**Expected behavior**
App should automatically add measurement and/or should explain how to create assisted user (and what is that).

**Additional context**
Probably it's just docs/UX issue.

NintyS, 2024-03-19:
I agree.

oliexdev, 2024-04-13:
well you need to connect to the scale first before you weight (click on the bluetooth button) and the assisted user is for assisted weighing (e.g. for babies or pets)

closed 2024-04-13 by oliexdev