oliexdev/openScale #1027: Unable to switch on some measurements

kir0ul, 2024-03-15:
**Describe the bug**
I'm unable to add some measurements (the body fat caliper and waist-hip ratio, other measurements seems to work fine) to the list of measurements, apparently I'm able to switch on the toggle buttons but the measurements names stay greyed out and are not added to the measurement list that can be logged. Also, nothing happens when the settings/cog button next to the toggle button gets pressed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Settings > Measurements`
2. Click on `Body fat caliper` or `Waist-hip ratio`, the toggle gets activated, but the name stay greyed out

Reproduced with: v2.5.2 (64)

**Expected behavior**
Click on the toggle button to add the measurement to the list of measurements that can be logged.

**Additional context**
Screenshot of the measurements settings:

[image]

Screenshot of the measurements list I can log:

[image]

kir0ul, 2024-04-02:
I tried uninstalling and reinstalling, but I still have the same issue.

oliexdev, 2024-04-13:
You can enable Wait-Hip measurement it if you enable the hip and waist measurment otherwise for the Wait-Hip ratio a measurement is missing, same for the caliper measurment, you need to enable all caliper measurement first.

closed 2024-04-13 by oliexdev