oliexdev/openScale #1034: BF500 support

ajayld, 2024-04-19:
**Scale name**
BF500 Beurer
https://play.google.com/store/apps/details?id=com.beurer.healthmanager&hl=en_IN&gl=US

Please add support for BF500 Beurer weighing scale.

_To support a new scale it is necessary to gather some information._

**Step 1: Read the general reverse engineer process**

- [ ] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex (male/female), body height, age, activity level
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
weight, date, water, muscle , fat etc.
```
_--> Attach the 1. HCI Snoop log file here <--_

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex (male/female), body height, age, activity level
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
weight, date, water, muscle , fat etc.
```

_--> Attach the 2. HCI Snoop log file here <--_

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex (male/female), body height, age, activity level
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
weight, date, water, muscle , fat etc.
```
_--> Attach the 3. HCI Snoop log file here <--_

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

_--> Attach the openScale debug log file here <--_

oliexdev, 2025-08-24:
I don't know, I don't own this scale but it seems somebody has implemented that. Try out the latest beta build --> https://github.com/oliexdev/openScale/releases/tag/beta-build and post the log file here maybe we see the error!?

user-mb, 2025-11-15:
Hi guys, anyone knows if this is working now?
thanks for anyone reading this.

(i would potentially buy this scale)

jmozmoz, 2025-12-06:
> Can confirm, this is how it worked for me before the v3.0 update. Now, the connection is lost and I cannot get it back.
Which version of the app worked for you? Did you build it yourself or can it be downloaded here as release?

Nezisi, 2026-01-03:
I think I will finish the implementation next week, maybe tomorrow.

@oliexdev pardon the ping, just want to make sure the PR is mergeable.

What the BF500 does is a bit weird as it seems - I debugged the Beurer Health App Pro, tried to trace down what's happening and comparing it to the existing implementation of [Pull Request 1110](https://github.com/oliexdev/openScale/pull/1110) - kudos to [ghtyrant](https://github.com/ghtyrant) 

```code:kotlin
class BeurerBF500Handler : StandardWeightProfileHandler() {

    private val SERVICE_UUID_CUSTOM = uuid16(0xFFFF)
    private val CHAR_UUID_ACTIVITY = uuid16(0xFFF2)
    private val CHAR_UUID_MEASUREMENT_REQUEST = uuid16(0xFFF4)

    override fun supportFor(device: ScannedDeviceInfo): DeviceSupport? {
        val name = device.name.lowercase(Locale.ROOT)
        if ("bf500" !in name) {
            return null
        }

        val caps = setOf(
            DeviceCapability.BODY_COMPOSITION,
            DeviceCapability.TIME_SYNC,
            DeviceCapability.USER_SYNC,
            DeviceCapability.BATTERY_LEVEL,
            DeviceCapability.LIVE_WEIGHT_STREAM
        )
        return DeviceSupport(
            displayName = "Beurer BF500",
            capabilities = caps,
            implemented = caps,
            linkMode = LinkMode.CONNECT_GATT
        )
    }

    /**
     * BF500-specific connection flow.
     *
     * 1. Set up UDS properties
     * 2. Check for saved mapping
     * 3. If saved: auto-consent using sendConsent()
     * 4. If new: register new user using sendRegisterNewUser()
     * 5. Write BF500 vendor data
     */
    override fun onConnected(user: ScaleUser) {
        logD("BF500: onConnected(userId=${user.id})")

        setupUDSProperties()

        val scaleIndex = findKnownScaleIndexForAppUser(user.id)
        if (scaleIndex == null) {
            // No saved mapping - register as new user
            val consent = Random.nextInt(0x0001, 0xFFFF)

            // StandardWeightProfileHandler => registration of user
            pendingConsentForNewUser = consent
            registeringNewUser = true
            pendingAppUserId = user.id

            logD("BF500: Registering new user with consent=$consent")
            sendRegisterNewUser(consent)
        } else {
            // Have saved mapping - auto-consent
            val consent = loadConsentForScaleIndex(scaleIndex)
            if (consent != -1) {
                logD("BF500: Auto-consenting with scaleIndex=$scaleIndex consent=$consent")
                sendConsent(scaleIndex, consent)
            } else {
                logD("BF500: Requesting consent for scaleIndex=$scaleIndex")
                requestScaleUserConsent(user.id, scaleIndex)
            }
[... 81 lines cut ...]
 
     override fun onNotification(characteristic: UUID, data: ByteArray, user: ScaleUser) {
@@ -731,7 +735,7 @@
 
     // ---- UDS command builders -------------------------------------------------
 
-    private fun sendRegisterNewUser(consentCode: Int) {
+    protected fun sendRegisterNewUser(consentCode: Int) {
         val payload = byteArrayOf(
             UDS_CP_REGISTER_NEW_USER.toByte(),
             (consentCode and 0xFF).toByte(),
@@ -741,7 +745,7 @@
         writeTo(SVC_USER_DATA, CHR_USER_CONTROL_POINT, payload)
     }
 
-    private fun sendConsent(scaleIndex: Int, consentCode: Int) {
+    protected fun sendConsent(scaleIndex: Int, consentCode: Int) {
         val payload = byteArrayOf(
             UDS_CP_CONSENT.toByte(),
             (scaleIndex and 0xFF).toByte(),
```

Hope its okay that I just copy pasted this here, but I think that makes it easier and hopefully lessens the burden regarding PR time. 

Thanks for all your efforts!

oliexdev, 2026-01-03:
@Nezisi thanks for asking, is it not possible to fix it in StandardBeurerSanitasHandler? What do you want to change, so it is working again?

Nezisi, 2026-01-03:
@oliexdev I could try to make it working. 

The issue seems to be the fallback case in StandardWeightProfileHandler.kt:
```
 // 2) No mapping → try to list users via UDS (may be unsupported)
 requestUdsListAllUsers()
```

My code does two things:
- removes the manual consent dialogs (as the scale doesn't show any consent at all - this took me a while to realize ;) )
- avoids the UDS list all users dialog

jmozmoz, 2026-01-03:
I have also looked at the bluetooth messages. Sorry, I do not really understand the whole protocol overhead, but what I have found is the messages reporting stored measurements on the scale:

They use the `handle`s `0x0020`, `0x0026`, and `0x0045`:

`0x0020`:
```
14 weigth weight YY YY mm dd HH MM SS  uu  BMI BMI Size Size
```
weigth: (two bytes) / 200 (kg)
YY: year (two bytes)
mm, dd, HH, MM, SS: (rest of date/time, one byte each)
uu: user id(?)
BMI: (two byte)/10 (-)
Size: (two bytes)/10 (cm)

`0x0026`:
```
153  3 BF BF ?? 26 Mu Mu XX XX YY YY ZZ ZZ 
```
BF: (two bytes)/10 body fat (%)
MU: (two bytes)/10 muskel (%)

I was not able to identify XX, YY, ZZ. Especially water percentage is missing which is displayed by the scale. I was able to fit water as a linear function of weight (kg) metabolic age (years) and XX, YY, ZZ.

`0x0045`:
```
12 255 255 VF VF MA
```
VF: (two bytes)/10 visceral fat (%)
MA: metabolic age

Two bytes in little-endian order. Constant values in all messages I logged are given explicitly.

In general is there somewhere an explanation how this whole bluetooth traffic works (what is characteristics, ...)?

To identify the messages I used the btatt filter in Wireshark.

In general, I did not find any existing code which uses a comparable encoding of the messages.

jmozmoz, 2026-01-07:
And bone mass is mass * (100 - Body fat percentage)/100 - "Soft Lean Mass in lb"/200

(which seems to be much too low compared to typical values, but this is what the scale calculates)