oliexdev/openScale #1038: Minor dark mode issue

alexanderadam, 2024-04-20:
**Describe the bug**

Buttons are barely visible in openScale Pro's dark mode

**To Reproduce**
Steps to reproduce the behaviour:

1. When creating a new user the datepicker for the birthday has black buttons in dark mode

[image]

**Expected behaviour**

The buttons should be properly visible instead

KaushalDoshi, 2024-05-06:
Hi, I want to fix this issue. Which color is expected instead of black for buttons?

alexanderadam, 2024-05-06:
I'm not the maintainer but most probably the same white like the calendar numbers and the pencil icon. Everything has the same colour except the buttons.

oliexdev, 2024-11-17:
should be finally fixed with the latest dev build.

closed 2024-11-17 by oliexdev