oliexdev/openScale #1042: Exception on App start

alexanderadam, 2024-04-21:
**Describe the bug**

I just got this exception when starting the app:

ˋˋˋ
Build version: 2.5.2-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2024-04-21 19:45:43 
Device: Samsung SM-T976B 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
android.app.ForegroundServiceStartNotAllowedException: startForegroundService() not allowed due to mAllowStartForeground false: service com.health.openscale.sync/.core.service.SyncService
	at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:54)
	at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:50)
	at android.os.Parcel.readParcelableInternal(Parcel.java:4787)
	at android.os.Parcel.readParcelable(Parcel.java:4755)
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3018)
	at android.os.Parcel.createException(Parcel.java:3007)
	at android.os.Parcel.readException(Parcel.java:2990)
	at android.os.Parcel.readException(Parcel.java:2932)
	at android.app.IActivityManager$Stub$Proxy.startService(IActivityManager.java:6234)
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1967)
	at android.app.ContextImpl.startForegroundService(ContextImpl.java:1933)
	at android.content.ContextWrapper.startForegroundService(ContextWrapper.java:839)
	at androidx.core.content.ContextCompat$Api26Impl.startForegroundService(ContextCompat.java:1091)
	at androidx.core.content.ContextCompat.startForegroundService(ContextCompat.java:749)
	at com.health.openscale.core.OpenScale.syncInsertMeasurement(OpenScale.java:717)
	at com.health.openscale.core.OpenScale.addScaleMeasurement(OpenScale.java:347)
	at com.health.openscale.gui.MainActivity$14.handleMessage(MainActivity.java:728)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8810)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)

ˋˋˋ

**To Reproduce**
Steps to reproduce the behavior:
1. Start the app

**Expected behavior**

Seeing no exception

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Digital365Staking, 2024-07-14:
The error message android.app.ForegroundServiceStartNotAllowedException: startForegroundService() not allowed due to mAllowStartForeground false typically indicates that the app is trying to start a foreground service while it’s in the background, which is not allowed in certain versions of Android](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows , https://stackoverflow.com/questions/73067939/start-foreground-service-after-notification-permission-was-disabled-causes-crash).

Here are a few steps you can take to address this issue:

Check the App Version: Make sure you’re using the latest version of the openScale app. If there’s a newer version available, consider updating the app as the issue might have been resolved in a later release.

[Check the Android Version: This issue is known to occur on Android 12 and above](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows)[1](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows). If you’re using one of these versions, you might encounter this problem.

Check the App’s Foreground Service: The error suggests that the app is trying to start a foreground service from the background. [This is not allowed in Android 12 and above unless the app is in the foreground or meets certain other criteria](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows)[1](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows). If possible, try to ensure that the app is in the foreground when starting the service.

Report the Issue: If none of the above solutions work, consider reporting the bug to the openScale community. [You can do this by creating a new issue in the openScale GitHub repository](https://stackoverflow.com/questions/70044393/fatal-android-12-exception-startforegroundservice-not-allowed-due-to-mallows)[3](https://stackoverflow.com/questions/44425584/context-startforegroundservice-did-not-then-call-service-startforeground). Be sure to include all the details you’ve provided here, as well as any additional information that might help the developers understand and reproduce the issue.

alexanderadam, 2024-07-14:
This sounds very like ChatGPT.
But then again, I'd expect ChatGPT to be able to read what's written in the post. :thinking: 

> Check the App Version

It literally says `Build version: 2.5.2-pro`

> [Check the Android Version

Again, it's literally written in the post (`OS version: Android 13 (SDK 33)`)

> Check the App’s Foreground Service: The error suggests that the app is trying to start a foreground service from the background. [This is not allowed in Android 12 and above unless the app is in the [...]

Apparently you are doing some double link encoding in your ChatGPT bot.
And this still happened on Android 13.

How terrible is this bot really?

> Report the Issue: If none of the above solutions work, consider reporting the bug to the openScale community. [You can do this by creating a new issue in the openScale GitHub [...]

Suggesting to open a GitHub issue on the GitHub issue that should be opened is definitely next level AI. :laughing:

oliexdev, 2025-05-10:
This problem might be resolved in the upcoming beta version (0.4) of openScale sync, which features a complete rewrite of the application and includes improvements to the synchronization logic.

We are currently running a closed beta test for this new version. If you'd like to help test and see if this issue is fixed, please consider joining the beta program. You can find more details and instructions on how to join in our pinned beta testing invitation issue here: https://github.com/oliexdev/openScale/issues/1124

closed 2025-05-10 by oliexdev