oliexdev/openScale #1043: openscale sync: specify reciever_export or reciever_not_exported flag.

Jackahack, 2024-04-26:
**Is your feature request related to a problem?.**
connecting  with openscale sync on android 14 phones to mqtt brokers will need to specify a certain flag (as per API level 34)
The situation right now when you connect is that app keeps crashing resulting in the following error: java.lang.SecurityException: com.health.openscale.sync: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts

what i found on the developer.adroid forum was the following

_"" Apps and services that target Android 14 (API level 34) or higher and use [context-registered receivers](https://developer.android.com/guide/components/broadcasts#context-registered-receivers) are required to specify a flag to indicate whether or not the receiver should be exported to all other apps on the device: either RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED, respectively. This requirement helps protect apps from security vulnerabilities by leveraging the [features for these receivers introduced in Android 13](https://developer.android.com/about/versions/13/features#runtime-receivers).
Exception for receivers that receive only system broadcasts
If your app is registering a receiver only for [system broadcasts](https://developer.android.com/guide/components/broadcasts#system-broadcasts) through Context#registerReceiver methods, such as [Context#registerReceiver()](https://developer.android.com/reference/android/content/Context#registerReceiver(android.content.BroadcastReceiver,%20android.content.IntentFilter)), then it shouldn't specify a flag when registering the receiver. ""_

see the link below
https://developer.android.com/about/versions/14/behavior-changes-14

the solution is to patch the app so it complies again with the api.   until then openscale sync is unusable for the latests android phones 
personally i dont know which flag is needed.

rd4p, 2024-06-23:
Hi, same problem here.
Mqtt option is not usable in the current state :x

oliexdev, 2025-05-10:
This problem might be resolved in the upcoming beta version (0.4) of openScale sync, which features a complete rewrite of the application and includes improvements to the synchronization logic.

We are currently running a closed beta test for this new version. If you'd like to help test and see if this issue is fixed, please consider joining the beta program. You can find more details and instructions on how to join in our pinned beta testing invitation issue here: https://github.com/oliexdev/openScale/issues/1124

closed 2025-05-10 by oliexdev