oliexdev/openScale #1050: Update Libra to openScale conversion script

abatula, 2024-06-04:
**Is your feature request related to a problem? Please describe.**
The existing Libra to openScale conversion [script in the Wiki](https://github.com/oliexdev/openScale/wiki/Useful-import-scripts#libra-csv-file----openscale-csv-file) didn't work in Python 3. Also, I had issues when importing my data because Libra exported lbs and it looks like openScale expected kg

**Describe the solution you'd like**
I updated the script to work for Python 3, and added an argument to tell the script it needs to convert from lbs to kg for the final file. https://gist.github.com/abatula/3bb6d83ce2d8ea1c906efbb55e933d37

**Describe alternatives you've considered**
Any feedback on the script and I'd be happy to change it. Or if there's a way to PR the wiki I can move this there.

**Additional context**
N/A

Digital365Staking, 2024-07-14:
The Python script you've provided already includes the functionality to convert weight from pounds (lbs) to kilograms (kg). The conversion is controlled by the `weight_scalar` variable, which is set to `0.453592` (the conversion factor from lbs to kg) when the script is run with 'lbs' as an argument. If the script is run without this argument, `weight_scalar` is set to `1`, meaning the weight values are assumed to be in kg and no conversion is performed.

Here's the relevant part of the script:

```python
if (len(sys.argv) > 2) and (sys.argv[2] == 'lbs'):
    print('Converting Libra data from lbs to kg')
    weight_scalar = 0.453592
else:
    print('Leaving Libra data as-is (kg)')
    weight_scalar = 1
```

And here's where the conversion is applied:

```python
weight_float = float(w[1]) * weight_scalar
```

So, if you want to convert the weight values from lbs to kg, you should run the script with 'lbs' as an argument, like this:

```bash
python libra_to_openscale.py Libra_2024-06-04.csv lbs
```

This will convert the weight values from lbs to kg before writing them to the output CSV file. If you run the script without 'lbs' as an argument, it will assume the weight values are already in kg and will not perform any conversion. 

I hope this helps! Let me know if you have any other questions.

abatula, 2024-07-19:
Ah, sorry if I wasn't clear. What I was suggesting was taking the script I included (the one you point out does have code to handle lbs conversion) and putting that in the wiki in place of [this one](https://github.com/oliexdev/openScale/wiki/Useful-import-scripts#libra-csv-file----openscale-csv-file) already there, which uses Python 2 and does not have the ability to convert from lbs. 

I would have liked to make a PR instead of open this issue, since I have a proposed solution/code to my problem, but I can't see a way to open a PR to change a wiki page.

oliexdev, 2024-11-17:
Thanks for your updated script I replaced it in the wiki https://github.com/oliexdev/openScale/wiki/Useful-import-scripts#libra-csv-file----openscale-csv-file

closed 2024-11-17 by oliexdev