oliexdev/openScale #1051: It is stating that the scale is not supported

Danjos012, 2024-06-20:
**Describe the bug*
The application does not recognize my scale as part of it, I would like it to include this balance too
D8:E7:2F:CF:1F:94
[openScale_2024-06-20_05-01.txt](https://github.com/user-attachments/files/15910936/openScale_2024-06-20_05-01.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): ???

**Expected behavior**
Add my scale 

**Additional context**
When I look for my scale it said it is not supported

**Debug log**
[openScale_2024-06-20_05-01.txt

Danjos012, 2024-06-20:
[openScale_2024-06-20_05-21.txt](https://github.com/user-attachments/files/15911077/openScale_2024-06-20_05-21.txt)
This file is more complete about searching for the scale on Bluetooth.

Danjos012, 2024-06-20:
BLE Scanner Chart Data
Device Name ,N/A
Mac Address ,D8:E7:2F:CF:1F:94

Time, RSSI
20-06-2024 05:46:35:556,-63
20-06-2024 05:46:37:556,-63
20-06-2024 05:46:39:557,-64
20-06-2024 05:46:41:556,-64
20-06-2024 05:46:43:556,-66
20-06-2024 05:46:45:557,-64
20-06-2024 05:46:47:560,-66

Metadados
10FFC048266B17700A0125D8E72FCF1F94

[openScale_2024-06-20_05-57.txt](https://github.com/user-attachments/files/15911460/openScale_2024-06-20_05-57.txt)
[openScale_2024-06-20_06-01.txt](https://github.com/user-attachments/files/15911461/openScale_2024-06-20_06-01.txt)
[image]

Danjos012, 2024-06-20:
It's not resolved yet, I accidentally closed it

Digital365Staking, 2024-07-14:
It seems like you're experiencing an issue where the openScale app is not recognizing your scale. This could be due to the scale not being supported by the openScale app. Here are a few steps you can take to troubleshoot this issue:

1. **Check the Supported Scales**: openScale supports a number of Bluetooth scales from various manufacturers. Make sure your scale is on the list of supported devices. If it's not, it might not work with the openScale app.

2. **Update the App**: Make sure you're using the latest version of the openScale app. If there's a newer version available, consider updating the app as the issue might have been resolved in a later release.

3. **Unpair and Re-pair the Scale**: If your scale is not showing up in openScale during the search at Settings->Bluetooth->Your Bluetooth scale, you must first unpair the scale with your smartphone in the Android Bluetooth settings or within an installed vendor app. Please also make sure that your scale is turned on and you granted the coarse location permission during the search.

4. **Check the Batteries**: Finally, check that the batteries of your scale are full enough.

5. **Add Support for New Scale**: If your scale is not supported, you could try to reverse engineer your scale by analyzing the Bluetooth 4.x protocol between your scale and the original vendor app. If you understand the protocol then it is not complicated to implement it into openScale.

If none of these steps resolve the issue, you might want to consider reaching out to the openScale community or support for more specific guidance. They might be able to provide more insight based on the specifics of your setup and the exact error messages you're seeing.

oliexdev, 2025-03-03:
try out the latest openScale version.