oliexdev/openScale #1052: Ignore out of range data seems not to be working

x00e, 2024-06-27:
Hi, my use case is the following: 

Both I and my partner are using the same smart scale. 

In openScale, I have setup a single user, for myself, with details including average weight. 
I have enabled Bluetooth > Ignore out of range data.

When using openScale to pull data from smartscale, all the readings are added my user, including those of my partner.
Between me and my partner there's a ~15Kg weight difference. I was expecting that since there's an obviously difference between the reading, only those within (some) deviation from my configured weight will be considered valid and added to DB. 

Using openScale v2.5.2(64) with a Xiaomi Mi Scale v.1
ngs -> About -> Save debug log to file) that is captured while reproducing the issue.

andrefcluz, 2024-07-08:
I'm having the same exact issue

Digital365Staking, 2024-07-14:
It seems like you're experiencing an issue where the openScale app is not correctly differentiating between the weight data from you and your partner. This could be due to the "Ignore out of range data" setting not working as expected. 

Here are a few suggestions that might help:

1. **Multiple Users**: Some users have reported that they were able to set up multiple users in the openScale app¹². You could try setting up a separate user for your partner in the app. This might help the app differentiate between the two sets of weight data.

2. **Update the App**: Make sure you're using the latest version of the openScale app. If there's a newer version available, consider updating the app as the issue might have been resolved in a later release³.

3. **Contact Support**: If none of the above solutions work, you might want to consider reaching out to the openScale community or support for more specific guidance. They might be able to provide more insight based on the specifics of your setup and the exact error messages you're seeing³.

Regarding the Xiaomi Mi Scale v.[1](https://www.reddit.com/r/Xiaomi/comments/vb6pfu/multiple_users_for_xiaomi_mi_body_composition/), it seems that the scale itself does not support multiple users¹³. The differentiation of users is done at the app level, not at the scale level. Therefore, the issue you're experiencing is likely related to the openScale app rather than the scale itself.

I hope this helps! Let me know if you have any other questions.

(1) Multiple users for Xiaomi Mi Body Composition Scale 2 : r/Xiaomi - Reddit. https://www.reddit.com/r/Xiaomi/comments/vb6pfu/multiple_users_for_xiaomi_mi_body_composition/.
([2](https://github.com/oliexdev/openScale/issues/1052)) Ignore out of range data seems not to be working #1052 - GitHub. https://github.com/oliexdev/openScale/issues/1052.
([3](https://www.reddit.com/r/Xiaomi/comments/ag7wqc/xiaomi_mi_smart_scale_2_multiple_users/)) Xiaomi Mi Smart Scale 2 - multiple users measurements : r/Xiaomi - Reddit. https://www.reddit.com/r/Xiaomi/comments/ag7wqc/xiaomi_mi_smart_scale_2_multiple_users/.
(4) Getting started with IBM Watson OpenScale (automated setup). https://developer.ibm.com/tutorials/getting-started-with-watson-openscale/.
(5) OpenScale Applications and Hookup Guide - SparkFun Learn. https://learn.sparkfun.com/tutorials/openscale-applications-and-hookup-guide/all.
(6) Issues · oliexdev/openScale · GitHub. https://github.com/oliexdev/openScale/issues.
(7) List Index Out of Range – Python Error [Solved] - freeCodeCamp.org. https://www.freecodecamp.org/news/list-index-out-of-range-python-error-solved/.
(8) “DataError: Integer out of range” on Postgres. What to do next?. https://medium.com/coffee-meets-bagel-engineering/dataerror-integer-out-of-range-on-postgres-what-to-do-next-d6fc71e935fa.
(9) std::out_of_range - cppreference.com. https://en.cppreference.com/w/cpp/error/out_of_range.
(10) Xiaomi Mi Smart Scale Unboxing, Setup & Demo | TechPP. https://www.youtube.com/watch?v=6H4rIc7bWR8.
(11) How to setup Mi Body Composition Scale 2 with Mi Fit app. https://www.youtube.com/watch?v=bZYpR47Ealc.
(12) How to setup Mi Smart Scale 2 with Mi Fit app. https://www.youtube.com/watch?v=l67_5222viw.
(13) How to Add Multiple Users to Xiaomi Mi Body Composition Scale 2 – Add .... https://www.youtube.com/watch?v=4tTkPY5acL8.
(14) undefined. https://www.hardreset.info/devices/xiaomi/xiaomi-mi-body-composition-scale-2/add-account/.