oliexdev/openScale #1054: Building source code with Android Studio

Pho6666, 2024-08-08:
I have been trying for hours to build the project with Android Studio so I can make some modifications to the project. I am still stuck.

What kind of configuration do I need to make in Android Studio? I have installed it on Windows 11 and entered the GitHub repo URL in the import from VCS menu.

Then in the Run/Debug Configurations menu it seems I have to create something, so I add an Android app. Do I have to select a specific name? It then forces me to select a module, but I can select none. I have no idea how to continue.

Any help would be really appreciated!

Pho6666, 2024-08-08:
Sorry, 'scale support' is the wrong tag here, but it is kind of a support request. Or maybe a bug, I don't know for sure.

closed 2024-08-09 by Pho6666

ayush-bharti, 2025-02-25:
@Pho6666 - did you figure out how to build it? Any guidance you can share?