oliexdev/openScale #1066: One Scale - Two Smartphones

udosw, 2024-09-05:
I recently got a Silvercrest SBF 77 scale and it works perfect with OpenScale (at least when I only used it with my smartphone alone).

Now I tried to link my wifes smartphone with the same scale as well. This turned out to be rather complicated.

First of all the scale always creates a new random pairing pin when a new device tries to connect. Ok, this is a one-time setup and it did work in the end.

But then no data was transferred to the 2nd smartphone at all. After some exprimenting it turned out, that I had to delete my wifes user account (in the openscale app) on my smartphone. And then to set her up as the only user in her own smartphone.

Also it seems that the scale cannot connect to two smartphone at the same time. 

I will wait and see how this works out during the next few days. I'd like to hear from others if they have experiences with this setup.

Blue-Sauce, 2025-02-07:
> Also it seems that the scale cannot connect to two smartphone at the same time.

This may be a limitation in the scale and there is nothing openScale can do, due to how bluetooth pairing is done.

Does the scale support multiple smartphone via the manufacturer's suggested app? If not, then it's likely the scale's firmware just doesn't support it.

Bluetooth first finds a paired device by the partner's bluetooth radio MAC address. Different devices should have different addresses. For the scale to support multiple address, it needs to store multiple MAC addresses & associated paring keys. Unfortunately, that is not a given. Note, even if you spoof your MAC to that of an already paired phone, it won't work unless without that phone's pairing key.

oliexdev, 2025-03-03:
nothing more to comment as Blue-Sauce already explained

closed 2025-03-03 by oliexdev