oliexdev/openScale #1069: BroadcastScale connects only once per start of the app

aircable, 2024-09-13:
I know you are working on the BroadcaseScale driver. We built a scale using that protocol.

May I ask for a feature that shows the weight continuously while the scale is reporting it? What I mean is that I can see the weight changing while stepping on the scale. When the scale is then stable it would be nice to have a button selecting the current measurement.

Right now, the app starts and takes the first packet it finds over air and uses it. There is no way to repeat the measurement after that. The only way is to close the app and restart it only when you are on the scale and be very quite.
It would make it more user friendly using this type of BroadcastScale.

Thanks!

sroemer, 2024-09-15:
I do not own a broadcast scale and don't know any details about its communication, anyway here some comments/ideas I have regarding this:

- Having a button to 'select' the current weight sounds not like a good idea to me.
I just imagine myself stepping on the scale, but then need the phone in my hand to push the button, which results in the weight being affected by the phone. 

- In my opinion a stable weight should be detected based on the messages sent by the scale or - if this is not possible - the final and stable weight should be detected based on identical values being received repeatedly for a number of times.

- Additionally a new measurement could be started when the weight drops back to 0 or drops by a certain amount.

oliexdev, 2025-03-03:
You could also step off the scale click on the bluetooth Button in openScale (it will disconnect to the scale) and if you click again it will try to connect again.

closed 2025-03-03 by oliexdev