oliexdev/openScale #1071: Backup/Import Feature Broken

dreamsyntax, 2024-09-20:
**Describe the bug**
Exporting the database and later attempting to import it does not transfer data successfully on version 2.5.2.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings > Backup> Export backup'
2. Delete the app and reinstall or clear all data.
3. Go to 'Settings > Backup > Import backup'
4. Choose openScale.db
5. Backup Import message appears, but all data is missing/lost.

**Expected behavior**
Import/Export feature can successfully transfer prior DB containing all recorded data.

MeikelLP, 2024-09-27:
Same for me but with a different message:

```
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2024-09-27 19:36:27 
Device: Fairphone FP5 
OS version: Android 14 (SDK 34) 
 
Stack trace:  
java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=1849680682, result=-1, data=Intent { dat=content://org.nextcloud.documents/... flg=0x1 }} to activity {com.health.openscale/com.health.openscale.gui.MainActivity}: android.os.NetworkOnMainThreadException
 at android.app.ActivityThread.deliverResults(ActivityThread.java:5528)
 at android.app.ActivityThread.handleSendResult(ActivityThread.java:5567)
 at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
 at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
 at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
 at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
 at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2444)
 at android.os.Handler.dispatchMessage(Handler.java:106)
 at android.os.Looper.loopOnce(Looper.java:205)
 at android.os.Looper.loop(Looper.java:294)
 at android.app.ActivityThread.main(ActivityThread.java:8223)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:977)
Caused by: android.os.NetworkOnMainThreadException
 at android.os.Parcel.createExceptionOrNull(Parcel.java:3067)
 at android.os.Parcel.createException(Parcel.java:3041)
 at android.os.Parcel.readException(Parcel.java:3024)
 at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:190)
 at android.database.DatabaseUtils.readExceptionWithFileNotFoundExceptionFromParcel(DatabaseUtils.java:153)
 at android.content.ContentProviderProxy.openTypedAssetFile(ContentProviderNative.java:814)
 at android.content.ContentResolver.openTypedAssetFileDescriptor(ContentResolver.java:2045)
 at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1860)
 at android.content.ContentResolver.openInputStream(ContentResolver.java:1530)
 at com.health.openscale.core.OpenScale.copyFile(OpenScale.java:488)
 at com.health.openscale.core.OpenScale.importDatabase(OpenScale.java:465)
 at com.health.openscale.gui.preferences.BackupPreferences.onActivityResult(BackupPreferences.java:242)
 at androidx.fragment.app.FragmentManager$8.onActivityResult(FragmentManager.java:2755)
 at androidx.fragment.app.FragmentManager$8.onActivityResult(FragmentManager.java:2735)
 at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
 at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
 at androidx.activity.ComponentActivity.onActivityResult(ComponentActivity.java:828)
 at androidx.fragment.app.FragmentActivity.onActivityResult(FragmentActivity.java:151)
 at com.health.openscale.gui.MainActivity.onActivityResult(MainActivity.java:996)
 at android.app.Activity.dispatchActivityResult(Activity.java:8943)
 at android.app.ActivityThread.deliverResults(ActivityThread.java:5521)
 ... 13 more
```

mike-lloyd03, 2024-10-15:
I was getting the same error as @MeikelLP (`NetworkOnMainThreadException`). I think because I was loading the file from my Nextcloud. After moving the file to my phone's local storage, there was no error but no data was imported.

MeikelLP, 2024-10-15:
Well I found have a more annoying issue:

The backup is empty. Sadly the backup function never seemed to work (for me) in the first place thus the import is failing...

dreamsyntax, 2024-11-26:
For anyone else, I ended up switching to trale since I was only tracking weight.
This one has a working backup feature if you constantly switch devices/reflash OS.
https://github.com/QuantumPhysique/trale

It does not have all the cool features of openScale, but figured I'd mention it for those who have simple use case.

damjang, 2024-12-08:
Confirm! Also for me the export/import don't work. Please can this be corrected?

oliexdev, 2025-03-03:
I cannot reproduce this error. I only found a bug that the imported backup will not shown instantly this should be fixed with https://github.com/oliexdev/openScale/commit/3e36fae2deca4126301106393afb8cffcda621c0

Where do you export your data? Android restrict it more recently, does it happen when you export it to "Downloads" folder?

pahlse, 2025-05-21:
I recently switched to a new phone. I installed openScale version 2.5.3 on both devices. I export the .db on my old phone, which works fine. Checking the db using sqlitebrowser shows no issues with the backup. When I adb push the file to my new phone and try to import it via the three dots on the Overview page I get the error: "Could not Import: header line is not valid: SQLite format 3@ ...". If I try to import from Settings > Backup > Import backup I get the message "Imported from /document/0677-AEF1:Download/openSca..." which didn't work the first try, but after spamming the import for a few time it worked.