oliexdev/openScale #1072: Add custom metrics support

iyesin, 2024-09-23:
**Is your feature request related to a problem? Please describe.**
To have all health records in one place, I would like to have ability to add customized (not so popular) metrics in app. For example: ketone levels, urine glucose level, water intake and some others.

**Describe the solution you'd like**
In metrics section I would like to see an option to add several named metrics, their limits (meaningful values) and units. Same metrics should be recorded in exported database. May be, it also worth reconsider export format: I'm fine with CSV/TSV but json and yaml handle dynamic fields better.

**Describe alternatives you've considered**
Obviously, there some alternatives from different scale vendors but most of them vendor-locked and cloud-based.
For now, I'm recording said metrics (ketone, glucose, water) in semi-random selected unused fields but this is weird and not very logical.

oliexdev, 2025-03-03:
please search for this request we have this question several times.

closed 2025-03-03 by oliexdev

iyesin, 2025-03-08:
Hey @oliexdev , thanks for your reply. What keywords should I use to search for previous discussions?