oliexdev/openScale #1078: Beuer BF 720 - wrong measurements

greatestradioshow, 2024-10-03:
**Describe the bug**
When using my Beuer BF 720 with openScale it tells me my body fat would be 27 %. When checking with Beuer's app, I get 8 %.
The value with openScale is definitely wrong.

**To Reproduce**
Steps to reproduce the behavior:
1. I do a test measure with the Beuer health manager to check my body fat value.
2. I perform a measure with the openScale app and compare the body fat value.
3. OpenScale recognizes the measure, but even when the weight is correct, the body fat is off.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 
Initially I installed 2.5.2 but even when downloading the dev-build I get the same result.
(I just installed the dev build over the normal one, in the settings I still see 2.5.2, not sure if I should have uninstalled it first).

**Expected behavior**
Body fat value should be correct.

**Additional context**
> I setup die Beuer app first and during the initial process a user was created. I did a first measure and everythig got logged (iPhone).
> I installed openScale on my Android, started it, and I got a pop up to select my user which i created before, but when trying to do so I'm asked for a PIN which I don't have. Is this normal?
> Therefore, I created a new user within openScale and this user is also visible on my BF 720. But even when I choose "male", my openScale user on the Beuer display shows me as "female". Is this a know behavior?

I already deleted all users and redid the setup, but the behavior stays the same.

**Debug log**
[openScale_2024-10-03_11-35.txt](https://github.com/user-attachments/files/17242375/openScale_2024-10-03_11-35.txt)

KlavsKlavsen, 2025-03-15:
press set on your weight and switch to the profile openscale created on the weight. My guess is that the height and age on the weight is NOT what you put into openscale.
I just found the same with my beuer f720.. I had put in 186cm and my date of birth - but in the weight it those 2 were wrong on my profile - which had a huge impact on my body fat percentage.

KlavsKlavsen, 2025-03-15:
I just tested with healthmanager pro (beuer) app - and it says a body fat of 14% - where openscale (after corrected profile on scale to correct height) says 25%.

So it seems openscale is for unknown reasons giving incorrect numbers :(

KlavsKlavsen, 2025-03-15:
I just looked at the weights numbers (when it switches to the user created by openscale) - and its the weight thats reporting those 25% bodyfat - so it must be simply some wrong setting getting set in the weight (as opposed to the user configured with beuer's app - where I get 14%).

So this seems to be due to openscale creating the role in the scale wrongly - and also - openscale does not notice that I changed settings for my profile on the scale itself. Health manager from beuer notices this, and asks if I want to overwrite those settings on the scale or not.. Openscale should do this check - that would catch such wrong settings in the weight - when app actually tries to read it back on each connect and compare to whats set in openscale user.

greatestradioshow, 2025-03-16:
As you could reproduce it, does this behavior look like a "bug"?

KlavsKlavsen, 2025-03-16:
It is clearly a bug and maybe a missing feature too:
1) The bug is that it creates the with very wrong values (leading to my very wrong measurements from the scale)
2) Its a missing feature, if openscale does NOT verify at connection time - that the details it has in its settings (my weight, birthday, gender etc.) matches whats in the profile it created on the weight - then thats a very important feature to have with this weight (as its editable via the weight). If it does check that - it clearly does so very poorly - and then its a bug :)

greatestradioshow, 2025-03-17:
It would be nice if this could be fixed by someone, as the 720 is mentioned at the main page as support scale.
Actually, this was the reason why I got this scale in the first place.

Kuki-netizen, 2025-06-10:
A bit late to the conversation, but I can report the same issue with my BF700. In the Beurer app I have a body fat of 18.4 and in OpenScale I'm getting around 22-24.