oliexdev/openScale #1079: Muscle trending up - red arrow

torantal, 2024-10-09:
I'm getting a red arrow for an up trending muscle mass, I guess this should be green?

Using openScale v2.5.2 (64)

[image]

oliexdev, 2025-03-03:
it depends on the "ideal" muscle mass in your age range, see the following paper
Muscle Reference: "Skeletal muscle mass and distribution in 468 men and women aged 18–88 yr" by IAN JANSSEN, STEVEN B. HEYMSFIELD, ZIMIAN WANG, and ROBERT ROS in J Appl Physiol89: 81–88, 2000

If your muscle mass "too high" regarding to the paper then the arrow will be red.

closed 2025-03-03 by oliexdev

torantal, 2025-03-03:
My muscle mass is definitely on the low side, I'm not into body building, just casual cardio